openbmc/openbmc #1460: Incorrect number of cpus in inventory

dkodihal, 2017-04-14:
On a single socket Witherspoon, I saw this -

    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Functional": 1, 
      "Manufacturer": "IBM", 
      "Model": "", 
      "PartNumber": "01HL322", 
      "Present": 1, 
      "PrettyName": "PROCESSOR MODULE", 
      "SerialNumber": "YA3933741575", 
      "Version": "10"
    }, 

    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1": {
      "Functional": 1, 
      "Present": 0, 
      "PrettyName": ""
    },

cpu1 was unexpected here. I ran dbus-monitor and saw a single property change -
string "xyz.openbmc_project.State.Decorator.OperationalStatus"
   array [
      dict entry(
         string "Functional"
         variant             boolean true
      )
   ]

This could be due to our sensor implementation, and/or host sensor data. @ratagupt , mind looking?

ratagupt, 2017-04-14:
I can understand that it could be due to sensor one but how "pretty name" property is coming in the output.
We do update the present and functional.

williamspatrick, 2017-04-14:
Pretty name comes from the inventory item itself. I assume something in inventory-manager enforces that every inventory item is at least an Inventory.Item?

dkodihal, 2017-04-14:
> I can understand that it could be due to sensor one but how "pretty name" property is coming in the output.

The inventory manager supplies defaults for properties the client did not fill.

ratagupt, 2017-04-21:
As per the mrw sensor id (a4) is for cpu1 functional sensor.
```
<globalSetting>
	<id>/sys-0/node-0/motherboard-0/proc_socket-1/module-0/p9_proc_m/cpu_func_sensor</id>
	<property>
	<id>IPMI_SENSOR_ID</id>
	<value>0xa4</value>
	</property>
</globalSetting> 
```

I could have seen that in the journal logs that host is sending the sensor data for this sensor id,hense on bmc we are creating the inventory object for the processor2.

```
Mar 30 11:47:20 witherspoon ipmid[810]: IPMI SET_SENSOR [0x5a]
Mar 30 11:47:20 witherspoon ipmid[810]: IPMI SET_SENSOR [0xa4]  >>>>>>>>cpu1 functional sensor
Mar 30 11:47:20 witherspoon ipmid[810]: IPMI SET_SENSOR [0x12]
```

Seems we need to ask from the host boot why they are sending the sensor info for the object which is not configured.

ratagupt, 2017-04-21:
More info 
here is the sensor info for proc0 and proc1 
```<globalSetting>
	<id>/sys-0/node-0/motherboard-0/proc_socket-0/module-0/p9_proc_m/cpu_func_sensor</id>
	<property>
	<id>IPMI_SENSOR_ID</id>
	<value>0x5a</value>
	</property>
</globalSetting>
<globalSetting>
	<id>/sys-0/node-0/motherboard-0/proc_socket-1/module-0/p9_proc_m/cpu_func_sensor</id>
	<property>
	<id>IPMI_SENSOR_ID</id>
	<value>0xa4</value>
	</property>
</globalSetting> 

```

williamspatrick, 2017-04-21:
Witherspoon is a two socket system so this makes sense right?

Only issue is that the processor is listed as functional but non-present. We must be interpreting something in that data different than Hostboot expects. 

Do they maybe expect us to ignore functional if non-present?  Maybe they set functional to avoid it looking like an error?

ratagupt, 2017-04-21:
w37 is a one socket correct?
if it is the case that all witherspoon is two socket system then why didn't host boot send the write fru for the other cpu.
I feel it should be consistent we should not get the sensor data for the processor for which we are not getting the fru data.

ratagupt, 2017-04-21:
Do they maybe expect us to ignore functional if non-present? Maybe they set functional to avoid it looking like an error?
Even I see it other way the possibility can be processor is present but not functional.

williamspatrick, 2017-04-21:
> w37 is a one socket correct?
> if it is the case that all witherspoon is two socket system then why didn't host boot send the write fru for the other cpu.

All Witherspoons are two sockets by board design.  w37 might only have one socket populated.

I think Hostboot is [mostly] doing the right thing here.  There is only one processor populated, so they cannot read the VPD EEPROM from a non-existent chip, so they cannot send us the 'writeFruData' command.  They do send us a "not present" indication, which is also 100% appropriate.

The only point of contention I see is why do they send "not present" + "functional".  This appears to be contradictory.

ratagupt, 2017-04-21:
It might happen that host has detected the presence but not able to read the VPD EEPROM then in that case we will not be having the write fru but if they send the sensor data then in that case it will be "Present" + "Non Functional".

williamspatrick, 2017-04-21:
Please look at #92 and open-power/hostboot#47.  I think we asked the Hostboot team to do the wrong thing a year ago.

ratagupt, 2017-04-22:
```Can you be more specific about where you are detecting the 'fault' setting? What bit(s) are you looking at? Which sensor type are you looking at (we have both functional and fault sensors)? For a non-present part, we deassert the present offset and assert the disabled offset. Is the disabled offset the problem?```

ratagupt, 2017-04-22:
That is exactly what we want and it is not happening or our understanding is incorrect.What I will do when I open a issue on host,I will paste the IPMI command and refer the (assert/dassert)bit value for the present offset and the functional offset.

williamspatrick, 2017-04-22:
Since we asked them to make the code change there was a similar defect on an AMI-based system. The bit here is actually listed as "disabled" in the IPMI spec. When they marked it non-present + disabled, a SEL was created for it being disabled. 

I don't think we can / should rely on any specific behavior from the host firmware. If a device is not-present it cannot be functional no matter what state the "disabled" bit is in.

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/3976 Ignore sensor(Memory/Proc) handling whose presence is false
Resolves #1460 Incorrect number of cpus in inventory

dkodihal, 2017-05-29:
This is being fixed by #1357.

closed 2017-05-29 by dkodihal