openbmc/openbmc #1461: REST GET request doesnt returns all object attributes. [Witherspoon]

gkeishin, 2017-04-17:
`Driver Info`
```
Build : https://openpower.xyz/job/openbmc-build/688/
BMC Version: v1.99.4-22-ga072ab2
PNOR: 
IBM-witherspoon-ibm-OP9_v1.15_1.14
        op-build-v1.15-75-g986c911-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-6a3a09c
        hostboot-a58f695
        linux-4.10.6-openpower1-pef4dc3d
        petitboot-v1.4.2-pe78e1ef
        witherspoon-xml-b208e4f
        occ-4141b5f
        hostboot-binaries-82b82cd
        capp-ucode-1bb7503

```

Enumerate shows correct inventory object properties data but accessing the object individual property returns wrong property data fields.

```
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9": {
        "BuildDate": "",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "0x2c80",
        "Model": "18ASF1G72PZ-2G6B1   ",
        "PartNumber": "",
        "Present": 1,
        "PrettyName": "0x0c",
        "SerialNumber": "0x156d5d8d",
        "Version": "0x31"
    }
}
```
```
$ rest_cmd -i xx.xx.xx.xx -o GET -u /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9
Executing: GET  /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9

/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9
{
    "Functional": 1,
    "Present": 1,
    "PrettyName": "0x0c"
}
```

Via DBUS the properties are showing correct
```
root@witherspoon:~# busctl call xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9 org.freedesktop.DBus.Properties GetAll s "xyz.openbmc_project.Inventory.Item"
a{sv} 2 "PrettyName" s "0x0c" "Present" b true
root@witherspoon:~# busctl call xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9 org.freedesktop.DBus.Properties GetAll s "xyz.openbmc_project.Inventory.Decorator.Asset"
a{sv} 5 "PartNumber" s "" "SerialNumber" s "0x156d5d8d" "Manufacturer" s "0x2c80" "BuildDate" s "" "Model" s "18ASF1G72PZ-2G6B1   "
root@witherspoon:~#
```

The same is seen in REST via automation, CURL.  and tried on multiple systems.

saqibkh, 2017-04-18:
According to Deepak:
this seems more like a REST server problem
We verified the inventory was fine via busctl commands.
Even the REST enumerate verb worked fine. It seems to be a problem only with the REST 'GET' verb.

rahulmah, 2017-05-02:
@saqibkh : Here is one more instance where we see that "Functional" property is not present for CPU0 in REST' Get request. But present in enumeration. 

```
bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0
{
  "data": {
    "BuildDate": "",
    "Cached": 0,
    "FieldReplaceable": 1,
    "Manufacturer": "IBM",                  <----   Functional field is not present with REST's Get.     
    "Model": "",
    "PartNumber": "01HL966",
    "Present": 1,
    "PrettyName": "PROCESSOR MODULE",
    "SerialNumber": "YA3933797382",
    "Version": "10"    <---- Functional field is missing here.
  },
  "message": "200 OK",
  "status": "ok"
```

```
 "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0": {
      "BuildDate": "",
      "Cached": 0,
      "FieldReplaceable": 1,
      "Functional": 1,              <--- Functional field is present in enumeration.
      "Manufacturer": "IBM",
      "Model": "",
      "PartNumber": "01HL966",
      "Present": 1,
      "PrettyName": "PROCESSOR MODULE",
      "SerialNumber": "YA3933797382",
      "Version": "10"
 ```

gkeishin, 2017-05-13:
Just a reminder its showing up on fan , we would need fix for this sooner.. our CT are breaking for quite sometime

```
Executing: GET  /xyz/openbmc_project/inventory/system/chassis/motherboard/fan2

/xyz/openbmc_project/inventory/system/chassis/motherboard/fan2
{
    "Present": 1,
    "PrettyName": "fan2"
}
```

```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan0": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan0"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan1": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan1"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan2": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan2"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan3": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan3"
    }
}
```

sivassrr, 2017-05-19:
With yesterday's driver 747 found it fixed. I believe we can enable this test. 
@gkeishin do you have any concern, still.

williamspatrick, 2017-05-19:
@dkodihal : Did we have any explicit code change that you are aware of to fix this?

gkeishin, 2017-05-19:
I m not aware of the fix.. I haven't seen this worked on builds I tested.. now build just came in.. I can verify it if there were any fixes..

gkeishin, 2017-05-19:
Nope.. its not yet fixed

```
On GET request
/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9
{
    "Functional": 1,
    "Present": 1,
    "PrettyName": "0x0c"
}

```

```
On enumeration
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9": {
        "BuildDate": "",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "0x2c80",
        "Model": "36ASF4G72PZ-2G6D1   ",
        "PartNumber": "",
        "Present": 1,
        "PrettyName": "0x0c",
        "SerialNumber": "0x15d27a35",
        "Version": "0x31"
    },
```

dkodihal, 2017-05-19:
> @dkodihal : Did we have any explicit code change that you are aware of to fix this?
No.

sivassrr, 2017-05-22:
yup, With the latest driver 751, its fail on another system.

ojayanth, 2017-07-27:
Looks like recent updates in rest server fixed this issue . Requested george to validate this

curl -c cjar -b cjar -k https://9.3.185.164/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9
{
  "data": {
    "BuildDate": "", 
    "Cached": 0, 
    "FieldReplaceable": 1, 
    "Functional": 1, 
    "Manufacturer": "0x2c80", 
    "Model": "36ASF4G72PZ-2G6D1   ", 
    "PartNumber": "", 
    "Present": 1, 
    "PrettyName": "0x0c", 
    "SerialNumber": "0x1434d91c", 
    "Version": "0x31"
  }, 
  "message": "200 OK", 
  "status": "ok"

dkodihal, 2017-07-27:
The issue was not always seen; @gkeishin and I saw this only for few items. Can you confirm this is fixed now @gkeishin?

gkeishin, 2017-07-27:
In the master,  latest https://openpower.xyz/job/openbmc-build/915/ builds it looks to be good. 
I need to check on the latter build to confirm the same.

geissonator, 2017-07-28:
So... we good here?  Do we know what fixed it?

gkeishin, 2017-07-31:
I see this good now.. what fixed this?  @dkodihal @ojayanth ?

dkodihal, 2017-08-07:
Can't pin-point right away since a bunch of changes went in a while bacl @gkeishin. Can we close this?

gkeishin, 2017-08-07:
Yeah we can close this one.

closed 2017-08-07 by gkeishin