openbmc/openbmc #1463: Sensor for host TPM requirement policy

geissonator, 2017-04-17:
Need to implement a sensor that will store a systems host TPM policy (whether it's required or not for boot).  Ideally the new sensor design is in place before implementing.  The hostboot team is looking for this ASAP though.

sivassrr, 2017-04-19:
@geissonator Hope we will get an option in openbmc gui as well, right.

mabaiocchi, 2017-04-24:
@geissonator who is assigned this work item so that I can keep him/her in-sync with the system owners adding this support?

geissonator, 2017-04-25:
@mabaiocchi - No assignee yet, the team decided we really needed to get the sensor refactor stories ( #778 epic) done prior to adding any new sensor functions.

Specifically we've got @dhruvibm looking into a sub-epic, #798 this sprint.

geissonator, 2017-05-08:
Per design meeting, this will come under the new settings design from @dkodihal #1465

geissonator, 2017-07-06:
Create the new dbus property like so - https://gerrit.openbmc-project.xyz/#/c/5033/
Then add in like so - https://gerrit.openbmc-project.xyz/#/c/4875/

devenrao, 2017-07-13:
- [ ] Create YAML interface file for TPM policy 
1. /xyz/openbmc_project/Control/Tpm/TpmPolicy.interface.yaml

- [ ] Specify the default value for the TPM policy at  meta-phosphor/common/recipes-phosphor/settings/phosphor-settings-defaults/defaults.yaml

devenrao, 2017-07-14:
Figured out the changes to be done. Plan to start coding. 14/07/2017

rfrandse, 2017-07-19:
<https://gerrit.openbmc-project.xyz/5378> Add in default for TPM enable/disable
Resolves  #1463 Sensor for host TPM requirement policy

mabaiocchi, 2017-07-28:
Hostboot is still looking for this support (3+ months now). Please let me know as soon as we can get this support on a Witherspoon BMC that we can use to test this functionality.  Thank you - Mike

closed 2017-08-02 by williamspatrick