openbmc/openbmc #1464: Create pre-power on systemd target with no timeout

anoo1, 2017-04-17:
There are some scenarios where we want to delay a system power on request, like during a pnor update, but delaying the power on causes the service to fail to start due to timing out.

Need to create a systemd target that's a pre-requisite to the power on services that doesn't have a timeout so that power on can be delayed without having to increase the timeout value to individual power on services.

geissonator, 2018-07-12:
We prevent the BMC from being rebooted during a BMC update but I don't think we do anything during a PNOR update. Not sure if we really need to spend the time on this one though since someone should naturally not boot a system if they're doing a code update of the host image.

geissonator, 2018-07-20:
Look into a long term solution for this, maybe use https://www.freedesktop.org/software/systemd/man/systemd-inhibit.html ?

anoo1, 2018-07-20:
> Look into a long term solution for this, maybe use https://www.freedesktop.org/software/systemd/man/systemd-inhibit.html ?

Yes, this would allow us to remove the current "mapper-subtree-remove" from the power service which waits for the BlocksTransition interface to not be present.
There could be a "code update" service which inhibits the power on service an potentially the BMC reboots.

closed 2019-10-27 by stale[bot]