openbmc/openbmc #1467: Select alternate SBE side on boot failures

geissonator, 2017-04-17:
There's a requirement the BMC do an SBE side switch if the reboot count hits 0.  Setting which SBE seeprom to boot from should just be a config bit somewhere but we'll need some code logic to do it.  Need to interlock with hostboot team on full design (i.e. they are responsible for updating and switching back at some point?)

geissonator, 2017-05-16:
- The design for P8 was for both the PNOR and SBE seeprom to boot from same "side" (golden vs. current).  For P9, the PNOR side is "floating" and can be programmed to not follow the SBE seeprom side.  What should our design be for P9?  Simplest is to keep them in synch again.  Need some discussion on how to pick PNOR side.  Need some discussion on "golden" concept.  SEEPROM selection is bit 17 in 0x2808 CFAM.

- Side A was golden and side B was the current, you boot from B first, try a second time on fail, generate error on fail, switch to golden, try 1 boot, if fails generate error and callout master proc.  If we go to A side, do we need to remember this?  Simpler design is to just try side B again after a host power off (or maybe BMC reset)?

- Where should this logic reside?  Putting within the phosphor-host-state-manager seems to fit.  Or we could put a service in the quiesce target that checks current PNOR side and does the side switch re-ipl if we can keep it simple enough.

spinler, 2017-05-16:
I think the actual setting of the register should be a procedure in openpower-proc-control, since it has all of the framework for that.

geissonator, 2017-05-16:
Proposed task list:
- [ ] Create new settingsd object to expose current seeprom/pnor side and allow user to set to A or B
- [ ] Add support in openpower-proc-control to switch SBE seeprom and PNOR side (single parameter taken as input, A or B)
- [ ] Create new phosphor-state-manager application that is called in service on quiesce state entry - Read boot count and current PNOR side, if 0 and B, then call new proc-control to switch side and then issue host reboot

rfrandse, 2017-05-26:
https://gerrit.openbmc-project.xyz/4265 Set SBE seeprom boot side
Resolves #1467 Select alternate SBE side on boot failures

closed 2017-06-03 by williamspatrick