openbmc/openbmc #1468: code-update : refactor

dkodihal, 2017-04-18:
Mainly address the comments here -

https://gerrit.openbmc-project.xyz/#/c/3647/2/version_image_manager.hpp@21

Talk to Gunnar because he may to get to some of this in the 04.05 sprint.

williamspatrick, 2017-04-18:
This sounds like a code review issue.  I don't think we should be opening a story to refactor brand new code.

anoo1, 2017-04-18:
Let's also address as part of this story that the openpower-version-host-software-manager core dumps if there's not a pnor image on the bmc.

dkodihal, 2017-04-19:
I don't have a better place to put this right away, but the image update location (/tmp/images or any other), should be created via a service I think. Right now I am doing that in the rest server, but it should be done via a service.

dkodihal, 2017-05-01:
Looks like Gunanr has most of this covered via review rework. Will keep this open but will un-assign Ratan.

dkodihal, 2017-07-21:
Closing this because the comments were addressed by Gunnar.

closed 2017-07-21 by dkodihal