openbmc/openbmc #1469: Error-log : enable open-powerd-bus  and application local errors

dkodihal, 2017-04-18:
- Enable processing of error YAML and error metadata YAML files under the openpower-dbus-interfaces. It should be possible for openpower-dbus errors to inherit metadata from phosphor-error interfaces.

- Enable applications to write app-local non-dbus errors. This will require elog-gen.py to be present in the SDK and installed via a native recipe, so that apps can add build rules. Need to decided if elog-gen.py suffices, or whether apps also need sdbus++ to run on error YAMLs

- Allow error YAMLs without associated metadata.

devenrao, 2017-04-20:
- [ ] Create a package group with all the repositories that have error yaml files 

- Modify the recipe file to create package group
- Modify the code in the depended repositories to include elog-error.hpp from the phosphor-logging

- [ ] Make the phostphor-logging depended on the package group created above

dkodihal, 2017-04-20:
More like -

- [ ] Via openpower-dbus-interfaces build, make sure error definitions get to a location where phosphor-logging can pick them

- [ ] Create package group to install application specific error yamls to a location where phosphor-logging can pick them. Have phosphor-logging depend on this package group.

- [ ] Convert an existing app that needs local errors to use this new infrastructure

- [ ] Change elog-gen.py to allow errors without metadata.

dkodihal, 2017-05-04:
Based on recent discussion, the following is required -

- The metadata types that elog-gen.py generates should not be sdbusplus exceptions. They should just be structures to hold error metadata.
- Error exceptions must be generated by running sdbus++, similar to how this happens for errors defined in the phosphor-dbus-interfaces repo. This means changes in the phosphor-logging Makefile to run sdbus++ on error yamls from the yaml location it will receive, and then install generated headers. If you can't run sdbus++ on a directory of error yaml's you may have to loop that directory in a script and call that script from the Makefile, or generate a separate Makefile for the sdbus++ invocation (like how it's done in phosphor-logging).

rfrandse, 2017-05-18:
https://gerrit.openbmc-project.xyz/#/c/4211/
(needs a Resolves openbmc/openbmc#1469 added to last commit)

closed 2017-06-08 by williamspatrick