openbmc/openbmc #1470: In-band IPMI SDR and FRU commands are not working as expected

PavamanSubramaniyam, 2017-04-18:
I have installed a P9 Witherspoon Hardware with Pegas 1.0 EA OS.
Then installed the ipmitool packages available from the Pegas 1.0 EA distro OS.

Executed the In-band IPMI commands on the machine from the HOST OS.

[root@ltc-wspoon4 ~]# ipmitool sdr info
Get SDR Repository Info command failed: Invalid command
[root@ltc-wspoon4 ~]# echo $?
1

[root@ltc-wspoon4 ~]# ipmitool fru list
FRU Device Description : Builtin FRU Device (ID 0)
 Device not present (Invalid command)

Err in cmd get sensor sdr info
[root@ltc-wspoon4 ~]# echo $?
1

[root@ltc-wspoon4 ~]# ipmitool -V
ipmitool version 1.8.18

[root@ltc-wspoon4 ~]# lsmod | grep ipmi
ipmi_powernv            6447  0
ipmi_devintf           13691  0
ipmi_msghandler        70388  2 ipmi_devintf,ipmi_powernv

The machine is installed with the BMC and PNOR images as follows:

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
[==================================================] 100%

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-22"
VERSION_ID="v1.99.4-22-ga072ab2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-22"
BUILD_ID="v1.99.4"

williamspatrick, 2017-04-18:
Not supported.

closed 2017-04-18 by williamspatrick

PavamanSubramaniyam, 2017-04-19:
Will the IPMI v2 compliant commands will be supported by OpenBMC