openbmc/openbmc #1471: IPMI SOL activate command fails

PavamanSubramaniyam, 2017-04-18:
I am trying to get to the SOL console of a P9 Witherspoon Open BMC hardware using Out of Band IPMI SOL command.

root@ltc-garri2:~# ipmitool -H 9.40.193.88 -I lanplus -P 0penBmc sol activate
Error activating SOL payload: Invalid command
root@ltc-garri2:~# ipmitool -H 9.40.193.88 -I lanplus -P 0penBmc sol deactivate
Error de-activating SOL payload: Invalid command

The P9 Witherspoon Hardware is installed with BMC and PNOR firmware levels as follows:

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-22"
VERSION_ID="v1.99.4-22-ga072ab2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-22"
BUILD_ID="v1.99.4"

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
[==================================================] 100%

root@ltc-garri2:~# ipmitool -H 9.40.193.88 -I lanplus -P 0penBmc chassis power status
Chassis Power is on

root@witherspoon:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10

The P9 Witherspoon Hardware is Powered ON and booted with Linux OS as can be seen above.

williamspatrick, 2017-04-18:
This is not yet supported.

closed 2017-04-18 by williamspatrick

PavamanSubramaniyam, 2017-04-19:
Could we know the OpenBMC release by which time this functionality will be supported.

williamspatrick, 2017-04-19:
https://github.com/openbmc/openbmc/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20sol