openbmc/openbmc #1473: SetTime throws error

trajeswaran, 2017-04-18:
In the MFG,  time mode is set to NTP. And then the time owner is changed to BMC and the mode is changed to MANUAL. Immediately after that, if time is set, then it results in error.

Please find below the sequence of curl command which when executed in a script results in error.  This seems to be a timing issue. 

Sequence of curl commands:
===================
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://192.168.72.233/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT https://192.168.72.233/org/openbmc/settings/host0/attr/time_mode -d "{\"data\": \"NTP\"}"

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://192.168.72.233/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT https://192.168.72.233/org/openbmc/settings/host0/attr/time_owner -d "{\"data\": \"BMC\"}"

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://192.168.72.233/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT https://192.168.72.233/org/openbmc/settings/host0/attr/time_mode -d "{\"data\": \"MANUAL\"}"

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://192.168.72.233/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://192.168.72.233/org/openbmc/TimeManager/action/SetTime -d "{\"data\": [ \"BMC\", \"2017-04-14 13:06:45\"]}"

Error Message
=========
{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle", 
      "    route, args = self.router.match(environ)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 823, in custom_router_match", 
      "    route.callback._setup(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 88, in _setup", 
      "    return self.setup(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 230, in setup", 
      "    request.route_data['method'] = self.find(path, method)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 223, in find", 
      "    m = self.find_method_on_bus(path, method, *items)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 259, in find_method_on_bus", 
      "    data = iface.Introspect()", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}

geissonator, 2017-04-18:
Please upload the full journal or run "ffdc" on the BMC and upload the generated tar file

trajeswaran, 2017-04-20:
Please find attached the FFDC.

[20170420034302183980_BMC_dmesg.txt.gz](https://github.com/openbmc/openbmc/files/941707/20170420034302183980_BMC_dmesg.txt.gz)
[20170420034302183980_BMC_general.txt.gz](https://github.com/openbmc/openbmc/files/941706/20170420034302183980_BMC_general.txt.gz)
[20170420034302183980_BMC_inventory.txt.gz](https://github.com/openbmc/openbmc/files/941710/20170420034302183980_BMC_inventory.txt.gz)
[20170420034302183980_BMC_journalctl.txt.gz](https://github.com/openbmc/openbmc/files/941709/20170420034302183980_BMC_journalctl.txt.gz)
[20170420034302183980_BMC_led.txt.gz](https://github.com/openbmc/openbmc/files/941708/20170420034302183980_BMC_led.txt.gz)
[20170420034302183980_BMC_meminfo.txt.gz](https://github.com/openbmc/openbmc/files/941711/20170420034302183980_BMC_meminfo.txt.gz)
[20170420034302183980_BMC_proc_list.txt.gz](https://github.com/openbmc/openbmc/files/941712/20170420034302183980_BMC_proc_list.txt.gz)
[20170420034302183980_BMC_procinfo.txt.gz](https://github.com/openbmc/openbmc/files/941713/20170420034302183980_BMC_procinfo.txt.gz)
[20170420034302183980_BMC_sensor_list.txt.gz](https://github.com/openbmc/openbmc/files/941714/20170420034302183980_BMC_sensor_list.txt.gz)

trajeswaran, 2017-04-20:
[20170420034302183980_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/941742/20170420034302183980_BMC_dmesg.txt)
[20170420034302183980_BMC_general.txt](https://github.com/openbmc/openbmc/files/941743/20170420034302183980_BMC_general.txt)
[20170420034302183980_BMC_inventory.txt](https://github.com/openbmc/openbmc/files/941744/20170420034302183980_BMC_inventory.txt)
[20170420034302183980_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/941745/20170420034302183980_BMC_journalctl.txt)
[20170420034302183980_BMC_led.txt](https://github.com/openbmc/openbmc/files/941746/20170420034302183980_BMC_led.txt)
[20170420034302183980_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/941747/20170420034302183980_BMC_meminfo.txt)
[20170420034302183980_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/941748/20170420034302183980_BMC_proc_list.txt)
[20170420034302183980_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/941749/20170420034302183980_BMC_procinfo.txt)
[20170420034302183980_BMC_sensor_list.txt](https://github.com/openbmc/openbmc/files/941750/20170420034302183980_BMC_sensor_list.txt)

williamspatrick, 2017-04-20:
```
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]: Traceback (most recent call last):
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 858, in _handle
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     route, args = self.router.match(environ)
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 826, in custom_router_match
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     route.callback._setup(**args)
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 89, in _setup
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     return self.setup(**kw)
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 232, in setup
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     request.route_data['method'] = self.find(path, method)
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 225, in find
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     m = self.find_method_on_bus(path, method, *items)
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 261, in find_method_on_bus
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     data = iface.Introspect()
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     **keywords)
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]:     message, timeout)
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]: DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Mar 30 13:59:52 witherspoon phosphor-gevent[1166]: ::8089:a076:8e91:f402%1157081884 - - [2017-03-30 13:59:52] "POST /org/openbmc/TimeManager/action/SetTime HTTP/1.1" 500 2032 25.223042
```

It does not appear that the "TimeManager" process is running.

williamspatrick, 2017-04-20:
```
root@witherspoon:~# mapper get-service /org/openbmc/TimeManager
:1.68
root@witherspoon:~# busctl tree :1.68 /org/openbmc/TimeManager --no-pager
Service :1.68:
└─/org
  └─/org/openbmc
    └─/org/openbmc/TimeManager

Service /org/openbmc/TimeManager:
Failed to introspect object / of service /org/openbmc/TimeManager: Invalid argument
No objects discovered.
```

There is a bug with TimeManager.

vishwabmc, 2017-07-17:
@williamspatrick I feel we can wait for the new Time Manager ?

vishwabmc, 2017-07-26:
@zahrens I dont feel its any priority. Not sure why its in sprint+1

vishwabmc, 2017-10-25:
@trajeswaran Please could you check on the latest time manager in xyz namespace and see if the issue is re-creatable ?

gtmills, 2018-07-05:
There is a new time manager in the xyz space, https://github.com/openbmc/phosphor-time-manager. Closing this. Please reopen if still an issue. thanks!

closed 2018-07-05 by gtmills