openbmc/openbmc #1474: From BMC Ready state, updating host0/attr/RequestedHostTransition to Host.Transition.On does not change Chassis transition state

trajeswaran, 2017-04-18:
At BMC Ready state, when I power on the host by updating RequestedHostTransition to Transition.On, the RequestedPowerTransitionState is not getting updated to Transition.On.  Unless this attribute is updated, it is not possible for the applications to know whether there has been any request for power on.

williamspatrick, 2017-04-18:
I don't see anything obvious here.  The code looks like it should have updated the property just fine.

```
Host::Transition Host::requestedHostTransition(Transition value)
{
    log<level::INFO>(
            "Host State transaction request",
            entry("REQUESTED_HOST_TRANSITION=%s",
                  convertForMessage(value).c_str()));

    Transition tranReq = value;
    if(value == server::Host::Transition::Reboot)
    {
        // On reboot requests we just need to do a off if we're on and
        // vice versa.  The handleSysStateChange() code above handles the
        // second part of the reboot
        if(this->server::Host::currentHostState() ==
            server::Host::HostState::Off)
        {
            tranReq = server::Host::Transition::On;
        }
        else
        {
            tranReq = server::Host::Transition::Off;
        }
    }

    executeTransition(tranReq);
    return server::Host::requestedHostTransition(value);   <---- 
}
```

williamspatrick, 2017-04-18:
@trajeswaran - We might need some evidence of what you're running and what you're seeing.  Is it possible the transition is happening fast enough that you are not observing a change in value?

geissonator, 2017-04-18:
I think the key here is that @trajeswaran want's to see the chassis requested state property get updated when he only sets the host requested state.  That code is not in place.  The chassis power current state property will get updated properly due to the systemd target file monitoring but not it's requested state property.

williamspatrick, 2017-04-18:
Oh.  I thought we said that the 'requested state' would always represent what the user intended to happen?  I don't see why we would automatically update some of those.

geissonator, 2017-04-18:
Yeah, he uses the requested states to tell the user what's going on.  He'd like to request a host power on, then see the chassis requested state go to powering on and inform the user of it.  (i.e. chassis is powering on, chassis is powered on, host is starting, then host is running).  I'll bring up in a design meeting to see if there's anything we want to do here.

geissonator, 2017-07-16:
There's general agreement this would be nice to have, but there's a lot of ramifications on other potential BMC implementations and what this may do.  The goal this sprint is to come to a final decision on this.

# STATUS
7/16 - No progress on this one yet, focused on last sprints story (code review comments)

geissonator, 2017-07-24:
One thought is to provide a mechanism for users to query the state of the targets responsible for the chassis and hosts.  That would provide a real time status on the chassis and host states but not impact the external interfaces into them.

gtmills, 2018-05-23:
@geissonator This defect is a year old, still a problem? Or can we close?

geissonator, 2018-05-23:
I'll let @trajeswaran have a say but I don't think we're going to fix this.  The state design is for separate objects for each (BMC, Chassis, Host).  The Requested portion of each is to reflect a direct user request change to that state. We don't want to go down a path where the state objects impact the other state Request fields.