openbmc/openbmc #1475: Witherspoon Powered Back On After Power-off Command

mabaiocchi, 2017-04-18:
I logged onto W51 this morning and saw that it was at the Ubuntu prompt.  I powered off, left the console, and when I came back it was back at the Ubuntu prompt.  

So I created the issue doing the following:

**Console showed this**:

mabaiocchi, 2017-04-18:
Here is the FFDC file created by running 'ffdc' on the OpenBmc cmdline 
[ffdc_2017-03-30_17-15-28.tar.gz](https://github.com/openbmc/openbmc/files/930349/ffdc_2017-03-30_17-15-28.tar.gz)

geissonator, 2017-04-18:
The last boot to happen in that trace shows we hit the host watchdog timeout and then because auto reboot was enabled (default) we power off and then back on again:
```
 9623 Mar 30 17:09:42 witherspoon systemd[1]: Reached target Host0 (Started).
 9624 Mar 30 17:09:42 witherspoon systemd[1]: Starting Start Watchdog0...
 9625 Mar 30 17:09:43 witherspoon host_watchdog.exe[735]: Setting watchdog poll interval to: 30000
 9626 Mar 30 17:09:43 witherspoon systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
 9627 Mar 30 17:09:43 witherspoon host_watchdog.exe[735]: Starting watchdog with poll interval: 30000
 9628 Mar 30 17:09:43 witherspoon systemd[1]: Started Start Watchdog0.
 9629 Mar 30 17:09:43 witherspoon systemd[1]: Reached target Chassis0 (Start).
 9630 Mar 30 17:09:43 witherspoon phosphor-host-state-manager[819]: Recieved signal that host is running
 9631 Mar 30 17:09:43 witherspoon phosphor-host-state-manager[819]: Change to Host State
 9632 Mar 30 17:10:13 witherspoon host_watchdog.exe[735]: Polling watchdog: 0
 9633 Mar 30 17:10:13 witherspoon chassis_control.py[740]: Watchdog Error, Going to quiesce
 9634 Mar 30 17:10:13 witherspoon systemd[1]: Reached target Quiesce Target.
 9635 Mar 30 17:10:13 witherspoon phosphor-host-state-manager[819]: Auto reboot enabled. Beginning reboot...
 9636 Mar 30 17:10:13 witherspoon phosphor-host-state-manager[819]: Host State transaction request
 9637 Mar 30 17:10:14 witherspoon systemd[1]: Stopped target Chassis0 (Start).
 9638 Mar 30 17:10:14 witherspoon systemd[1]: Started Stop Power0.
```

Must be something in this autoreboot path that's not playing well with hostboot.  I know there's an SBE bug out here that's causing the host watchdog to hit during boot, but not sure how this could happen when the user has requested a power off.  I'll dig around a bit.  Must be some combo of an auto reboot mixed in with a power off.

If you don't need the auto reboots then running this before using a system will probably fix it:
busctl call org.openbmc.settings.Host /org/openbmc/settings/host0 org.freedesktop.DBus.Properties Set sss org.openbmc.settings.Host auto_reboot no

williamspatrick, 2017-04-18:
Why was it powering on at all though?  Mike is claiming he didn't do it himself.

geissonator, 2017-04-19:
Bill recreated today, and based on his description I came up with this theory
- Hostboot team member boots system up, hits a fail, does some debug
- During that debug, the 2 minute host watchdog expires and attempts an autoreboot
- Due to the 45s mbox delay, we sit at powered on/running state for a while 
- Hostboot dev issues a poweroff
- The autoreboot code is currently not aware of the poweroff request so it still does the autoreboot

This doesn't fit with Mike's description but his journal did indicate we hit the host watchdog and were doing an auto reboot so not sure if that was a after affect or what.  The only code path where we autoreboot the system is if the host watchdog hits or if we hit a checkstop (maybe Mike hit that?)

mabaiocchi, 2017-04-19:
I don't know if I hit a checkstop as I did not actually try any commands at the Ubuntu prompt.  I just did the 'obmcutil poweroff' command from the other window.  I'm sure you can recreate this easily on w51

amboar, 2017-04-20:
As mbox got a mention above, [the bump to mboxd](https://gerrit.openbmc-project.xyz/#/c/3738/) should remove any interference on its behalf. There's no-longer a need to delay. There is a [patch for skeleton](https://gerrit.openbmc-project.xyz/#/c/2766/4) that probably also needs submitting, but in my tests yesterday the first change was enough to immediately poweron the system after a succesful poweroff.

johnhcwang, 2017-04-24:
Also recreated on zaius (b5520b14b1d71f6365712017a3a2e647d4744965).
The host states in Quiesced after boot to petitboot and RequestedHostTransition is Reboot (base on above mention, it should be SBE bug that's causing the host watchdog to hit).
```
root@zaius:~# busctl call xyz.openbmc_project.State.Host /xyz/openbmc_project/state/host0 org.freedesktop.DBus.Properties GetAll s xyz.openbmc_project.State.Host
a{sv} 2 "RequestedHostTransition" s "xyz.openbmc_project.State.Host.Transition.Reboot" "CurrentHostState" s "xyz.openbmc_project.State.Host.HostState.Quiesced"
```
Then I transfer host to Off. The system went back to ON soon.
The RequestedHostTransition doesn't be changed to Off and hit https://github.com/openbmc/phosphor-state-manager/blob/master/host_state_manager.cpp#L307.
```
root@zaius:~# busctl call xyz.openbmc_project.State.Host /xyz/openbmc_project/state/host0 org.freedesktop.DBus.Properties Set ssv xyz.openbmc_project.State.Host RequestedHostTransition s xyz.openbmc_project.State.Host.Transition.Off

root@zaius:~# busctl call xyz.openbmc_project.State.Host /xyz/openbmc_project/state/host0 org.freedesktop.DBus.Properties GetAll s xyz.openbmc_project.State.Host
a{sv} 2 "RequestedHostTransition" s "xyz.openbmc_project.State.Host.Transition.Reboot" "CurrentHostState" s "xyz.openbmc_project.State.Host.HostState.Off"

root@zaius:~# busctl call xyz.openbmc_project.State.Host /xyz/openbmc_project/state/host0 org.freedesktop.DBus.Properties GetAll s xyz.openbmc_project.State.Host
a{sv} 2 "RequestedHostTransition" s "xyz.openbmc_project.State.Host.Transition.Reboot" "CurrentHostState" s "xyz.openbmc_project.State.Host.HostState.Quiesced"
```
```
Apr 24 10:00:57 zaius phosphor-chassis-state-manager[926]: Received signal that power OFF is complete
Apr 24 10:00:57 zaius phosphor-chassis-state-manager[926]: Change to Chassis Power State
Apr 24 10:00:57 zaius phosphor-host-state-manager[924]: Received signal that host is off
Apr 24 10:00:57 zaius phosphor-host-state-manager[924]: Change to Host State
Apr 24 10:00:57 zaius phosphor-host-state-manager[924]: Reached intermediate state, going to next
Apr 24 10:00:57 zaius power_control.exe[929]: PowerControl: setting power up SOFTWARE_PGOOD to 1
Apr 24 10:00:57 zaius power_control.exe[929]: PowerControl: setting power up BMC_POWER_UP to 1
Apr 24 10:00:57 zaius power_control.exe[929]: PowerControl asserted latch BMC_UCD_LATCH_LE
Apr 24 10:00:57 zaius systemd[1]: Started Start Power0.
Apr 24 10:00:57 zaius timemanager[960]:  PGOOD has changed..
Apr 24 10:00:57 zaius system_manager.py[890]: Running System State: HOST_POWERING_ON
```

geissonator, 2017-05-11:
Anyone seen this in recent builds?  We removed the 45s sleep on power off's which if my theory is correct, should pretty much have gotten rid of this issue.

mine260309, 2017-05-11:
I did not test Witherspoon but I can see this on Palmetto.
If auto reboot is enabled, and it fails to boot, eventually host state becomes Quiesced state, while chassis power is on.
```
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
``` 
In this case, issue `obmcutil poweroff` will power off the chassis, and because of auto-reboot, it will re-poweron the host.
```
May 11 02:31:39 palmetto systemd[1]: Started Stop Power0.
May 11 02:31:39 palmetto systemd[1]: Starting Stop Watchdog0...
May 11 02:31:39 palmetto systemd[1]: Starting Wait for Power0 to turn off...
May 11 02:31:39 palmetto systemd[1]: Stopped target Start Host0.
May 11 02:31:39 palmetto systemd[1]: Stopped target Chassis0 (Power On).
May 11 02:31:39 palmetto systemd[1]: Stopping Phosphor Fan Presence Tach Daemon...
May 11 02:31:39 palmetto systemd[1]: Stopped Start Watchdog0.
May 11 02:31:39 palmetto systemd[1]: Stopped target Host0 (Started).
May 11 02:31:39 palmetto systemd[1]: Stopped target Host0 (Starting).
May 11 02:31:39 palmetto systemd[1]: Stopped Start Host0.
May 11 02:31:39 palmetto systemd[1]: Stopped target Quiesce Target.
May 11 02:31:39 palmetto systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
May 11 02:31:39 palmetto systemd[1]: Stopped Phosphor Fan Presence Tach Daemon.
May 11 02:31:39 palmetto systemd[1]: Stopped target Power0 (On).
May 11 02:31:39 palmetto systemd[1]: Stopped target Power0 On.
May 11 02:31:40 palmetto systemd[1]: Stopped Start Power0.
May 11 02:31:40 palmetto systemd[1]: Stopped Wait for Power0 to turn on.
May 11 02:31:40 palmetto host_watchdog.exe[671]: Stopping watchdog
May 11 02:31:40 palmetto power_control.exe[755]: PowerControl: setting power up POWER_PIN to 1
May 11 02:31:40 palmetto timemanager[798]:  PGOOD has changed..
May 11 02:31:40 palmetto system_manager.py[710]: Running System State: HOST_POWERING_OFF
May 11 02:31:41 palmetto systemd[1]: Started Stop Watchdog0.
May 11 02:31:41 palmetto phosphor-host-state-manager[753]: Auto reboot enabled. Beginning reboot...
May 11 02:31:41 palmetto phosphor-host-state-manager[753]: Host State transaction request
May 11 02:31:42 palmetto system_manager.py[710]: Running System State: HOST_POWERED_OFF
May 11 02:31:42 palmetto sensor_manager2.py[679]: Deleting OCC device
May 11 02:31:42 palmetto timemanager[798]:  User Settings have changed..
May 11 02:31:42 palmetto timemanager[798]:  PGOOD has changed..
May 11 02:31:42 palmetto timemanager[798]: Changing time settings allowed now
May 11 02:31:42 palmetto power_control.exe[755]: PowerControl: pgood: 0, setting reset USB_RESET to 0
May 11 02:31:42 palmetto systemd[1]: Started Wait for Power0 to turn off.
May 11 02:31:42 palmetto systemd[1]: Reached target Power0 Off.
May 11 02:31:42 palmetto systemd[1]: Reached target Chassis0 (Power Off).
May 11 02:31:42 palmetto systemd[1]: Reached target Stop Host0.
May 11 02:31:42 palmetto phosphor-chassis-state-manager[757]: Received signal that power OFF is complete
May 11 02:31:42 palmetto phosphor-chassis-state-manager[757]: Change to Chassis Power State
May 11 02:31:42 palmetto phosphor-host-state-manager[753]: Received signal that host is off
May 11 02:31:42 palmetto phosphor-host-state-manager[753]: Change to Host State
May 11 02:31:42 palmetto phosphor-host-state-manager[753]: Reached intermediate state, going to next
May 11 02:31:43 palmetto systemd[1]: Stopped target Power0 Off.
May 11 02:31:43 palmetto systemd[1]: Stopped target Stop Host0.
May 11 02:31:43 palmetto systemd[1]: Stopped Stop Watchdog0.
May 11 02:31:43 palmetto systemd[1]: Stopped target Chassis0 (Power Off).
May 11 02:31:43 palmetto systemd[1]: Stopped Wait for Power0 to turn off.
May 11 02:31:43 palmetto systemd[1]: Stopped Stop Power0.
May 11 02:31:43 palmetto systemd[1]: Starting Wait for Power0 to turn on...
May 11 02:31:43 palmetto systemd[1]: Starting Start Power0...
```

This is technically the case the issueing "power-off" makes the host "power-on"

geissonator, 2017-05-11:
Ahhh, that helps, I may see a bug there.  We get signals when we enter and exit a target.  I think we're getting the exit signal for quiesce and doing the policy of auto rebooting.  Josh, I think we just need to add the logic we did elsewhere to be sure we're entering the quiesce state target before doing the auto reboot logic.

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/3949 Check Quiesce State before issuing auto reboot
Resolves #1475 Witherspoon Powered Back On After Power-off Command

closed 2017-05-18 by williamspatrick