openbmc/openbmc #1476: [Barreleye v1.99.4-34] SquashFS fails during update

yadron, 2017-04-18:
This bug is similar to #1443 and see "parent bug" #1452 also: 

```
t@barreleye:/run/initramfs# rm ./image-bmc
root@barreleye:/run/initramfs# mv ./image-bmc.2 ./image-bmc  
root@barreleye:/run/initramfs# ./update
update:
/dev/mtdblock5 /run/initramfs/ro squashfs ro,relatime 0 0
ERROR: Device mtd5 is mounted, image-bmc is busy.
root@barreleye:/run/initramfs# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     248224         0    248224   0% /dev
tmpfs                   249504     42740    206764  17% /run
/dev/mtdblock5           14720     14720         0 100% /run/initramfs/ro
/dev/mtdblock6            4096       456      3640  11% /run/initramfs/rw
cow                       4096       456      3640  11% /
tmpfs                   249504         0    249504   0% /dev/shm
tmpfs                   249504         0    249504   0% /sys/fs/cgroup
tmpfs                   249504        44    249460   0% /tmp
tmpfs                   249504        24    249480   0% /var/volatile
root@barreleye:/run/initramfs# umount /dev/mtdblock5
root@barreleye:/run/initramfs# umount /dev/mtdblock6
root@barreleye:/run/initramfs# ./update
update:
Updating bmc...
Erasing block: 512/512 (100%)
Writing kb: 32768/32768 (100%)
Verifying kb: 32768/32768 (100%)
root@barreleye:/run/initramfs# reboot
SQUASHFS error: xz decompression failed, data probably corrupt
SQUASHFS error: squashfs_read_data failed to read block 0xc81fc
SQUASHFS error: Unable to read data cache entry [c81fc]
SQUASHFS error: Unable to read page, block c81fc, size bcb8
SQUASHFS error: Unable to read data cache entry [c81fc]
SQUASHFS error: Unable to read page, block c81fc, size bcb8
SQUASHFS error: Unable to read data cache entry [c81fc]
SQUASHFS error: Unable to read page, block c81fc, size bcb8
SQUASHFS error: Unable to read data cache entry [c81fc]
SQUASHFS error: Unable to read page, block c81fc, size bcb8
SQUASHFS error: Unable to read data cache entry [c81fc]
SQUASHFS error: Unable to read page, block c81fc, size bcb8
SQUASHFS error: Unable to read data cache entry [c81fc]
SQUASHFS error: Unable to read page, block c81fc, size bcb8
-sh: /sbin/reboot: Input/output error
```

mdmillerii, 2017-04-18:
First of all I can see that the system was not in `update` mode but running from the mounted file system.

The update script correctly detected this and deferred the update.

however, the user proceeded to try to fix the issue, but did so incorrectly and the kernel was still using the file system because the reference from the  `cow` file system on `/` was still active.  Unfortunately without a kernel change there is no way to detect the `mtd` partition is busy before doing the flash.

If the update script complains the file system is mounted the user must either cancel the update (by removing the file) or let the update proceed by running the `reboot` command.

williamspatrick, 2017-04-18:
You need to be in update mode before attempting an update.  Please see the following documentation:

https://github.com/openbmc/docs/blob/master/code-update.md#update-from-the-openbmc-shell

Otherwise you are effectively writing the raw flash out from underneath a running file system.  The probability of this ever working are fairly low.

closed 2017-04-18 by williamspatrick