openbmc/openbmc #1478: mboxd: Handle multiple mount types

anoo1, 2017-04-18:


amboar, 2017-04-19:
As mentioned in #1479 can we partly handle these (e.g. ro/rw) with union mounts?

anoo1, 2017-04-21:
We could for ro/rw, just the challenge is that we have at least 3 different partitions (4 including the preserved one), with different purposes.
@amboar maybe you have some ideas to see if/how we could implement the overlay. In summary, we have:
ro and rw partitions - self explanatory
patch (tmp) - This is a directory somewhere in the BMC (/tmp?) where the user would copy a specific pnor partition and the BMC will then use it instead of the copy in rw/ro. This allows mfg or dev to 'patch' the pnor.
preserved - These are the pnor partitions marked as preserved that don't get updated, like nvram.

amboar, 2017-04-24:
I can understand the motivation to enable manufacturing to patch certain partitions, but is it necessary to split this concept from the RW partition? Given we have the original partitions in an RO filesystem is there a strong reason to do something beyond "reset" the state by removing some or all of the files in the RW filesystem?

If there is a strong desire to have something like a "patch" partition, can we use systemd to configure the mounts?

My thought is we could do something like this in the non-patch case:

```
+--------------+                  +----------------+                        +----------------+
|              |                  |                |                        |                |
| /mnt/fw/rwfs +--------+         |                |                        |                |
|              |        |         |                |                        |                |
+--------------+        +---------+  /mnt/fw/lower +------------------------>  /mnt/fw/root  |
|              |        |         |                |                        |                |
| /mnt/fw/rofs +--------+         |                |                        |                |
|              |                  |                |                        |                |
+--------------+     overlayfs    +----------------+       bind-mount       +----------------+
```

And for the patch case:

```
+--------------+                                                            +----------------+
|              |                                                            |                |
| /mnt/fw/patch+-------------------------------------------------+          |                |
|              |                                                 |          |                |
+--------------+                  +-----------------+            |          |                |
|              |                  |                 |            +---------->  /mnt/fw/root  |
| /mnt/fw/rwfs +--------+         |                 |            |          |                |
|              |        |         |                 |            |          |                |
+--------------+        +---------+  /mnt/fw/lower  +------------+          |                |
|              |        |         |                 |                       |                |
| /mnt/fw/rofs +--------+         |                 |                       |                |
|              |                  |                 |                       |                |
+--------------+    overlayfs     +-----------------+       overlayfs       +----------------+
```

Where `/mnt/fw/patch` is either just a raw directory or a tmpfs mountpoint. I haven't dug in to figure it out, but I expect systemd units can be used to juggle the mounts? So manufacturing can do something like 

```
# # Put system in "manufacturing patch mode"
# systemd start fw-mfg
# cp foo /mnt/fw/root
# obmcutil poweron
# ...
# # Reset system to "normal" mode
# systemd start fw-system
```

By marking the units with "Conflicts=..." it should implement the switch.

There is a minor twist in that the name of the unit must match the mount path for mount units. Because of this we would have to use a oneshot-style service unit that invokes `mount` to configure the bind mount (in the normal case) or upper overlayfs (for the patch case).

Also there is the complication of the "preserved" concept. Instead of a separate partition can we push the responsibility into the application handling the flashing process? For instance when we write an image to the BMC flash we can backup whitelisted files in the rwfs, and this is a similar problem. One way or another either the mbox daemon or the flashing process will need to be provided with an explicit list of partitions that are to be "preserved", so I'm not introducing a new problem but shifting where the problem is solved.

If all of this is achievable it means the modifications to mboxd are restricted to using the TOC to map flash offsets to files and file-offsets in `/mnt/fw/root`. Further, with this approach, `default.target` should depend on the `fw-system` unit outlined above. This means we won't default to providing the host with "patch" partitions if they are persisted on the BMC `rwfs` (i.e. if a tmpfs isn't used).

dkodihal, 2017-07-20:
This was implemented with #1479. #1551 will address pnor patching.

closed 2017-07-20 by dkodihal