openbmc/openbmc #1479: mboxd: Handle write back to written sections

anoo1, 2017-04-18:


amboar, 2017-04-19:
Can we union mount the RW filesystem over the RO filesystem? That way we can leave these tricks to the kernel.

ratagupt, 2017-05-08:
- [ ] Find the window for the given offset.

- [ ] If it is not there then load it into memory.

- [ ] write it to the partition.

- [ ] Create unit test.

amboar, 2017-05-09:
> Find the window for the given offset.

This is unnecessary. We do not currently support multiple active windows. Instead, any dirty or flush commands sent apply to the current active window. Thus there should be no searching involved.

> If it is not there then load it into memory.

It must be in memory by definition, from the protocol spec. Please read over it carefully.

> write it to the partition.

Hopefully there are only two minor issues here:

1. Identify which partition file you need to write according to the precedence hierarchy as apparently there are issues with my overlay ideas. So, ro (which you can ignore for the purpose of writes), rw, or patch (and I think there was another option?).
2. Calculate the offset within the partition from the dirty offset for the window. If your windows are sized to the partition and you map window offset 0 to partition offset 0, then this should be a no-op as the dirty offset with respect to the partition file might simply be the offset into the window.

rfrandse, 2017-05-31:
<https://gerrit.openbmc-project.xyz/4511> test: write the vpnor file(flash)
Resolves  #1479 mboxd: Handle write back to written sections

closed 2017-06-29 by williamspatrick