openbmc/openbmc #1480: Mismatch in BMC current side Firmware images shown in Petitboot System Information UI

PavamanSubramaniyam, 2017-04-19:
There is a mismatch in the BMC current side Firmware images shown in Petitboot System Information UI and OpenBMC ssh login.

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-34"
VERSION_ID="v1.99.4-34-gdd8a547"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-34"
BUILD_ID="v1.99.4"

 Petitboot System Information
 ──────────────────────────────────────────────────────────────────────────────

 BMC current side:
        Device ID: 0x0
        Device Rev: 0x80
        Firmware version: 1.99.66560
        IPMI version: 2

PavamanSubramaniyam, 2017-04-19:
Looks like an issue with the Auxiliary portion of the device ID response. The IPMI spec marks the format of this as vendor-specific, and so far all existing users have just set this as a 4 byte unsigned number.

From the raw response:
> # ipmitool raw 0x06 0x01
>  00 80 01 99 02 8d 41 a7 00 4f 42 00 04 01 00
The auxiliary portion: 
> 00 04 01 00

It looks like OpenBMC may be setting that in BCD. Can we get someone from the OpenBMC team to confirm what format this is in? *Preferably* this can match existing users so we don't need a special case just for OpenBMC.

williamspatrick, 2017-04-19:
The spec says:

```
When the vendor-specific definition is not known, generic utilities should display each
byte as 2-digit hexadecimal numbers, with byte 13 displayed first as the most-
significant byte.
```

If you are displaying it as decimal you're not following the spec.

Now, there appears to be a problem with our implementation in that it has a buffer overflow.  The 01 should not have been in there.  It also might only properly support v1.x tags and not v1.x.y tags.  We should probably revisit how we are doing our versioning anyhow.

williamspatrick, 2017-04-19:
@tomjoseph83 - Buffer overflow:

```
            memcpy(&dev_id.aux, rev.d, 4);
```

Memcpy'ing 4 bytes from a 2 byte field (rev.d).  That means we always end up with random garbage in the last two bytes of the aux field.

tomjoseph83, 2017-05-04:
@williamspatrick

```
uint16_t d[2];
```
d is a 4 byte field.

tomjoseph83, 2017-08-05:
@williamspatrick Is there anything to be addressed with this issue?

tomjoseph83, 2017-09-01:
# **STATUS**

09/01
I didn't get a chance to debug this issue.

closed 2017-12-05 by geissonator