openbmc/openbmc #1482: barreleye system going into Quiesce state after code update

prkatti1, 2017-04-20:
barreleye system going into Quiesce state after code update

Version details:
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-42"
VERSION_ID="v1.99.4-42-g360a31b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-42"
BUILD_ID="v1.99.4"

Linux barreleye 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Wed Apr 19 22:24:01 UTC 2017 armv5tejl GNU/Linux

pnor info:
IBM-barreleye-ibm-OP8_v1.15_2.4
        op-build-v1.15-96-gc612094
        buildroot-2017.02-4-gc28573e
        skiboot-5.4.3
        hostboot-695bd89
        linux-4.10.6-openpower1-p7669d59
        petitboot-v1.4.2-p1493d08
        barreleye-xml-253622f
        occ-301b535
        hostboot-binaries-5c78ef3
        capp-ucode-1bb7503

sivassrr, 2017-05-04:
@prkatti1 shall we close this? With the new PNOR it is working fine, right.

prkatti1, 2017-05-04:
Yes @sivassrr , I am closing it.

closed 2017-05-04 by williamspatrick

yadron, 2017-05-10:
Why this bug closed?

I've just updated Barreleye to latest and still always have

```
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```

_(so I cannot boot host)_

The Barreleye itself is native _(I didn't do any hardware changes and currently I have one CPU out of two and few DIMMs of memory there)_. This Barreleye Host booted well yesterday when I had 1.99.0 version, but still problems updating to latest.

The ver of OpenBMC is 100% original too now:
```

cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-21"
VERSION_ID="v1.99.5-21-g2933545"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-21"
BUILD_ID="v1.99.5"
```

and both I've just updated to latest PNOR from
https://openpower.xyz/job/openpower-op-build/lastStableBuild/distro=ubuntu,target=barreleye/
, so all binaries as it without any modifications.

yadron, 2017-05-11:
As additional info, looks like something going wrong waiting for PGOOD:

```
systemctl list-jobs
  JOB UNIT                                 TYPE  STATE  
50080 obmc-start-watchdog@0.service        start waiting
50063 obmc-host-starting@0.target          start waiting
49870 phosphor-fan-presence-tach@0.service start waiting
50060 op-start-host@0.service              start waiting
49869 obmc-host-start@0.target             start waiting
50068 obmc-power-start@0.target            start waiting
50065 obmc-chassis-poweron@0.target        start waiting
50066 [[0;1;39mop-wait-power-on@0.service          [[0m start [[0;1;39mrunning[[0m
50085 pcie-slot-detect@0.service           start waiting
49918 obmc-power-on@0.target               start waiting
50082 obmc-host-started@0.target           start waiting

[[0;1;39m11 jobs listed[[0m.
```

(both
```
root@barreleye:/lib/systemd/system# obmcutil power      
pgood = 0
state = 1
pgood_timeout = 0
```
)

, and after "op-wait-power-on" timeout, it goes to Quiesce state.