openbmc/openbmc #1483: While performing ISTEP, ISTEP 16. 2 shows value as 0.00515

PavamanSubramaniyam, 2017-04-20:
When we do obmutil poweron from the BMC ssh login, we get the ISTEPs initiated as follows.

--== Welcome to Hostboot  ==--

  4.09152|secure|SecureROM invalid - skipping functionality
 11.31620|ISTEP  6. 5
 12.04685|ISTEP  6. 6
 12.04782|ISTEP  6. 7
 13.64377|HWAS|PRESENT> DIMM[03]=FF00CF0000000000
 13.64380|HWAS|PRESENT> Proc[05]=8000000000000000
 13.88922|ISTEP  6. 8
 13.89046|SECURE|Security Access Bit> 0x0000000000000000
 13.89047|SECURE|Secure Mode Disable (via Jumper)> 0x8000000000000000
 13.89063|ISTEP  6. 9
 14.58826|ISTEP  6.10
 14.63799|ISTEP  6.11
 14.63900|ISTEP  6.12
 14.83616|ISTEP  7. 2
 16.82715|ISTEP  7. 3
 17.40869|ISTEP  7. 4
 21.00519|ISTEP  7. 5
 21.08009|ISTEP  8. 1
 21.35039|ISTEP  8. 2
 21.37592|ISTEP  8. 3
 21.37757|ISTEP  8. 4
 21.52417|ISTEP  8. 5
 21.54874|ISTEP  8. 6
 21.55103|ISTEP  8. 7
 21.65006|ISTEP  8. 8
 21.69864|ISTEP  8. 9
 21.82096|ISTEP  8.10
 21.82276|ISTEP  8.11
 21.82328|ISTEP  8.12
 22.14183|ISTEP  8.13
 22.16738|ISTEP  8.14
 22.16871|ISTEP  8.15
 22.17041|ISTEP  8.16
 22.41596|ISTEP  8.17
 22.53297|ISTEP  8.18
 22.53430|ISTEP  8.19
 22.56116|ISTEP  8.20
 22.56350|ISTEP  9. 1
 22.70972|ISTEP  9. 2
 22.73521|ISTEP  9. 3
 22.73664|ISTEP  9. 4
 22.76205|ISTEP  9. 5
 22.76363|ISTEP  9. 6
 22.86219|ISTEP  9. 7
 22.91252|ISTEP  9. 8
 22.93875|ISTEP 10. 1
 23.42568|ISTEP 10. 2
 25.86147|ISTEP 10. 3
 25.86368|ISTEP 11. 1
 25.88994|ISTEP 12. 1
 25.93959|ISTEP 13. 1
 26.09255|ISTEP 13. 2
 26.21756|ISTEP 13. 3
 26.30241|ISTEP 13. 4
 26.31544|ISTEP 13. 6
 26.48852|ISTEP 13. 7
 26.49063|ISTEP 13. 8
 28.22448|ISTEP 13. 9
 28.66412|ISTEP 13.10
 29.79703|ISTEP 13.11
 30.44592|ISTEP 13.12
 30.44838|ISTEP 13.13
 30.58785|ISTEP 14. 1
215.00022|ISTEP 14. 2
215.10047|ISTEP 14. 3
215.17792|ISTEP 14. 4
215.18057|ISTEP 14. 5
215.42714|ISTEP 14. 6
215.47799|ISTEP 14. 7
215.48336|ISTEP 15. 1
217.87618|ISTEP 15. 2
217.90252|ISTEP 15. 3
217.93696|ISTEP 15. 4
218.08584|ISTEP 16. 1
  0.00515|ISTEP 16. 2
219.36772|ISTEP 16. 3
219.42700|ISTEP 16. 4
219.51029|ISTEP 16. 5
219.52982|ISTEP 18.11
220.19450|ISTEP 18.12
220.16949|ISTEP 20. 1
222.20675|ISTEP 20. 2
223.77216|ISTEP 21. 1
243.35094|ISTEP 21. 2
243.34523|ISTEP 21. 3
[  244.005605960,5] SkiBoot skiboot-5.5.0 starting...
[  244.005609097,5] initial console log level: memory 7, driver 5
[  244.005611889,6] CPU: P9 generation processor(max 4 threads/core)
[  244.005614517,7] CPU: Boot CPU PIR is 0x004c PVR is 0x004e0100
[  244.005617230,7] CPU: Initial max PIR set to 0x7fff
[  244.006828237,5] OPAL table: 0x300d2840 .. 0x300d2d00, branch table: 0x30002000
[  244.006834169,7] Parsing HDAT...
[  244.006835907,5] SPIRA-S found.
[  244.006839998,6] BMC #0: HW version 3, SW version 2, chip DD1.0
[  244.006844847,6] SP Family is ibm,ast2500,openbmc
[  244.006852200,7] LPC: IOPATH chip id = 0
[  244.006854092,7] LPC: FW BAR       = f0000000
[  244.006856133,7] LPC: MEM BAR      = e0000000
[  244.006858208,7] LPC: IO BAR       = d0010000
[  244.006860240,7] LPC: Internal BAR = c0012000

As can be seen above, the values displayed are incremental while performing ISTEP's.
But when we look closer, we see that all ISTEP's shows the ranges in increasing order of 216...217...218, but suddenly for ISTEP 16.2 it shows 0.00515, and after that again shows incremental values 219....220... for ISTEP 16.3 etc.

215.47799|ISTEP 14. 7
215.48336|ISTEP 15. 1
217.87618|ISTEP 15. 2
217.90252|ISTEP 15. 3
217.93696|ISTEP 15. 4
218.08584|ISTEP 16. 1
  0.00515|ISTEP 16. 2         <<========
219.36772|ISTEP 16. 3
219.42700|ISTEP 16. 4
219.51029|ISTEP 16. 5
219.52982|ISTEP 18.11
220.19450|ISTEP 18.12

PavamanSubramaniyam, 2017-04-20:
This happens on both Witherspoon and Zaius OpenBMC Hardware's installed with the latest upstream OpenBMC Firmware and PNOR images.

williamspatrick, 2017-04-20:
This is not a BMC issue. Please take this up with the Hostboot team.  We have no control over the P9 time base register.

closed 2017-04-20 by williamspatrick