openbmc/openbmc #1489: phosphor-settingsd : move users to new settings interfaces

dkodihal, 2017-04-20:
Move users to new settings interfaces.

dkodihal, 2017-06-15:
- [ ] auto-reboot : move state manager code to use new settings object
- [ ] boot-flags : move host ipmid to new settings object
- [ ] power-policy : move host ipmid to new settings object
- [ ] time-owner and time-mode : move time manager code to new setting objects

dkodihal, 2017-07-13:
13 July : No progress. Working on PNOR bringup with new mboxd code. Delivery of that needs to happen in the next few days. Being a bringup activity, there's no story for this. Also spent a good amount of time reviewing code from almost everyone in the squad, and helped squad members with direction for the new sprint's stories and task breakdown.

rfrandse, 2017-07-25:
<https://gerrit.openbmc-project.xyz/5480> restriction_mode: switch to new settings API
Resolves  #1489 phosphor-settingsd : move users to new settings interfaces

closed 2017-08-01 by williamspatrick