openbmc/openbmc #1496: ObjectMapper [Feature Request]: Get service "well-known" names

mine260309, 2017-04-24:
The methods provided by `org.openbmc.objectmapper.ObjectMapper` are to get the `DBUS unique connection name(s)` begin with `:`, e.g. `:1.11`.
Such names will change if a service is restarted.

As an application/service, when it tries to call other service's method, currently we use `GetObject()` to find the unique name, and then make the call, which consumes two IPCs and it applies to each attempt to call a method, because the unique name is not guaranteed valid if the callee service is restarted.

I would like to use the "well-known" name of the service during the method call, so that an application/service can find other service's "well-known" name, save it, and use it for all subsequent method calls. This saves 1 IPC for each method call after the first call, because the "well-known" name is guaranteed valid even if the callee service is restarted.

So this feature request asks ObjectMapper to add methods to get other services' "well-known" names.

E.g.
```
### Current GetObject returns unique name
busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetObject sas "/org/openbmc/control/power0" 0
a{sas} 1 ":1.43" 2 "org.openbmc.Control" "org.openbmc.control.Power"   # It returns ":1.43"

### If we have GetObjectName to return "well-known" name
busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetObjectName s "/org/openbmc/control/power0"
# It shall return "org.openbmc.control.Power"
s "org.openbmc.control.Power"
```

williamspatrick, 2017-04-24:
@bradbishop -thoughts on this?  We've talked about it before.

rfrandse, 2017-10-26:
<https://gerrit.openbmc-project.xyz/7579> server: Serve well known bus names
Resolves  #1496 ObjectMapper [Feature Request]: Get service "well-known" names

closed 2017-12-15 by geissonator