openbmc/openbmc #1497: OpenBMC ASM GUI: No changes after click on icon "Server Power'

sivassrr, 2017-04-24:
BMC Level:

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-52"
VERSION_ID="v1.99.4-52-gb5520b1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-52"
BUILD_ID="v1.99.4"
root@witherspoon:~# cat /etc/timestamp
20170421193051
root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.15_1.28
	op-build-v1.15-131-gda28b74-dirty
	buildroot-2017.02-4-gc28573e
	skiboot-5.5.0
	hostboot-de49800
	linux-4.10.7-openpower1-pd408964
	petitboot-v1.4.2-p7985f01
	witherspoon-xml-07ef29d
	occ-e1889e5
	hostboot-binaries-d1b1312
	capp-ucode-1bb7503
	sbe-c9fea2e-p3dffe09
[==================================================] 100%

Test Ran: Click on icon "Server Power"  (Found system reached quiesce state, So had to click the icon "Server Power"). 
 Result: Nothing had come, Earlier on click, used to get all power operations related buttons
Based on this, we had automation our TCs

Expected Result:
Click on "Server Power" icon corresponding power on / off buttons expected.

sivassrr, 2017-04-24:
[image]

sivassrr, 2017-04-24:
[image]

iftekharuli, 2017-04-24:
Hi Sivas. I have updated the test site fixing the issue. Please let me know if it is working fine now :)

sivassrr, 2017-04-25:
Great, I checked it and found working @iftekharuli . Thank you.

sivassrr, 2017-05-17:
Hope this changes whatever you made will be there in the official code which you folks are check-in. Shall I close this. And I can verify once the official code available in the driver.

closed 2017-07-27 by williamspatrick