openbmc/openbmc #1498: OpenBMC ASM GUI: On click Icon "Server health" - no changes

sivassrr, 2017-04-24:
BMC Level:

root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-52"
VERSION_ID="v1.99.4-52-gb5520b1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-52"
BUILD_ID="v1.99.4"
root@witherspoon:# cat /etc/timestamp
20170421193051
root@witherspoon:# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[ ] 0%IBM-witherspoon-ibm-OP9_v1.15_1.28
op-build-v1.15-131-gda28b74-dirty
buildroot-2017.02-4-gc28573e
skiboot-5.5.0
hostboot-de49800
linux-4.10.7-openpower1-pd408964
petitboot-v1.4.2-p7985f01
witherspoon-xml-07ef29d
occ-e1889e5
hostboot-binaries-d1b1312
capp-ucode-1bb7503
sbe-c9fea2e-p3dffe09
[==================================================] 100%

Test Ran: Click on icon "Server health" 
Result: No changes happened. Expected to see url https://openbmc-test.mybluemix.net/#/server-health

Expected Result:
Click on "Server health" icon expected to see https://openbmc-test.mybluemix.net/#/server-health

This was working earlier.

iftekharuli, 2017-04-24:
@sivassrr  Server health should take you to log page for now as we decided. We might revisit the design discussion later but as of right now, Server health will take you to the first item in navigation which is event logs. Jack can give you more info on this @alfordja

sivassrr, 2017-09-19:
Will double check when it comes to official driver. So closing it now.

closed 2017-09-19 by sivassrr