openbmc/openbmc #1500: Evaluate conditions in order to create the correct fan zones

spinler, 2017-04-24:
This is an extension of story #1445 (Create fan control app and set initial fan speeds), which needs #370 (Fan control - Air cool vs Water cool policies) to be done first.

The fan zones present in a system may depend on the actual system config, such as if the system uses water cooling.  So, the zone config yaml specifies some conditions that must be true for the zones to apply, and the code will need to evaluate those conditions at runtime before loading the zones.

The plan is to just implement a condition that checks dbus properties, though leaving the syntax open to be able to add other types of conditions in the future.

gtmills, 2017-05-17:
- [ ] Add Conditions to control types
- [ ]  Add condition logic in control manager

rfrandse, 2017-06-06:
<https://gerrit.openbmc-project.xyz/4631> Check for zone number and condition
Resolves  #1500 Evaluate conditions in order to create the correct fan zones

closed 2017-06-26 by williamspatrick