openbmc/openbmc #1501: [Palmetto] System is not reaching BMC Ready state

swe12345, 2017-04-25:
palm303  is not reaching BMC Ready state even after reboot
```
root@palmetto:~# uptime
 05:27:18 up 6 min,  load average: 0.10, 0.57, 0.38
```
```
root@palmetto:~# obmcutil state
Traceback (most recent call last):
  File "/usr/sbin/obmcutil", line 216, in <module>
    obj = bus.get_object(bus_name, obj_path)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name xyz.openbmc_project.State.BMC was not provided by any .service files
```
```
root@palmetto:~# cat /etc/timestamp
20170420160242

```
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-42"
VERSION_ID="v1.99.4-42-g360a31b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-42"
BUILD_ID="v1.99.4"
```
[dmesg_25apr.txt](https://github.com/openbmc/openbmc/files/953984/dmesg_25apr.txt)
[journal_25apr.txt](https://github.com/openbmc/openbmc/files/953985/journal_25apr.txt)

williamspatrick, 2017-04-25:
systemctl list-jobs is the best command to help debug this situation.

gkeishin, 2017-05-22:
I looked at the log, this system with a custom build BMC kernel leftover by Dev.  So this is not on the master build or release build. So closing this issue.

closed 2017-05-22 by gkeishin