openbmc/openbmc #1506: Generic Bash script based dump collection infrastructure implementation

ojayanth, 2017-04-25:
— Add support for Dump summary collection
  — Add support Base scrips used in all types of Dump.

ojayanth, 2017-07-27:
Splitting this story into two One pointer.
   - This story will cover Dreport base infrastructure for command line enablement.

Created new story for  #2037  base script support.

ojayanth, 2017-07-27:
- [ ] dReport command line tool interface enablement:
     Enable all  must to have command line options backend support.
- [ ] Infrastructure for  use case specific dump.
     This include dummy functions to call use case specific dump.
- [ ] Enable Summary log support.
      Define format and and enable all required  summary info ( Details in the mailing list )

- [ ] Add support for Error handling .
    Include logging and return errors to caller.

ojayanth, 2017-07-28:
Scope:   Enable User initiate and application core dump support in dreport.
   This includes enabling Dump manager must to have features to trigger dump and packaging. 
    Dump data collection is based on re-using ffdc  data collection  functions.

rfrandse, 2017-08-09:
<https://gerrit.openbmc-project.xyz/5892> Enabled dreport based dump data collection
Resolves  #1506 dReport base infrastructure.

closed 2017-08-22 by williamspatrick