openbmc/openbmc #1508: REST-API: Add File read command.

ojayanth, 2017-04-25:
Develop REST API for File read .This can be used for reading  Dump File from BMC remotely.

ojayanth, 2017-06-15:
Assumption : where user will give the Path of the Dump.

- [ ] Implement the route handler for  the read request.

- [ ] Copy the Dump file .

dkodihal, 2017-06-15:
- [ ] Define a rest route for downloading a dump. The route would look like "download/dump/<dump_id>".

- [ ] Map the dump id to the dump file. This can be done for example if the dump file is placed under foo/dumps/<dump_id>/<dump_file>.

- [ ] Serve the dump file using the Bottle's static_file method. This, given a file path, prepares an HTTP response with details such as size, mime type and transfers the file.

rfrandse, 2017-06-29:
<https://gerrit.openbmc-project.xyz/5023> rest_dbus: allow route 'download/dump/<dumpid>;
Resolves  #1508 REST-API: Add File read command.

closed 2017-07-07 by williamspatrick