openbmc/openbmc #1510: BMC Dump: Add support for User initiated Dump

ojayanth, 2017-04-25:
- Add required Bash script for user use initiated dump.
  - Enable this support in Dump Service

dkodihal, 2017-05-15:
This story will cover enabling a user initiated dump via appropriate interface - internally just call the FFDC script for now, which can be later changed to suit whatever dump content is needed. Reducing to 1 point.

ojayanth, 2017-05-17:
- [ ] Add yaml support for create() and reporDump() interfaces.

- [ ] Implementation of create() and reporDump() d-bus interfaces

- [ ] Add infrastructure code support  to enable  manager class  for initiating existing FFDC bash script for user-inited dump use case

- [ ] Add infrastructure  code support  to enable  create class .

rfrandse, 2017-06-14:
<https://gerrit.openbmc-project.xyz/4787> Add support for user initated BMC Dump
Resolves  #1510 BMC Dump: Add infrasture support for User initiated Dump

closed 2017-07-19 by williamspatrick