openbmc/openbmc #1517: Create a RestFul interface to upload a ssh key to the BMC

whowutwut, 2017-04-25:
We'd like to have the ability to upload a ssh key to the BMC to ~/ssh/authorized_keys to provide passwordless ssh to the BMCs.

bradbishop, 2017-06-29:
Can we provide this in 2.0?  If not XCAT will have to write an expect script to upload the keys over SSH.

bradbishop, 2017-06-29:
@williamspatrick @geissonator ^

mdmillerii, 2017-06-29:
upload a file using fimrware update tftp transfer and move to authorized_keys ?

williamspatrick, 2017-07-06:
I think realistically we're not going to be able to get this done for phase 5.  Seems more likely to be a 2.1 feature, which is probably too late for xCAT?

whowutwut, 2017-07-06:
@williamspatrick Thanks for the info.  This should be OK.  Being proactive, I had @gurevichmark work around this by using expect scripts to send the keys. ( https://github.com/xcat2/xcat-core/pull/3376 ) once the REST function is available, we can modify this code and use it.  At least we can close some of our gaps.  

So go ahead and move this item down on your priority list.  

Do we know at what point certain data would persist on the BMC after a flash? I would assume that the `/root/.ssh` directory won't be erased on firmware flashing.

williamspatrick, 2017-07-06:
When we implement this, we'll figure out a way to make sure it is preserved through a code update.  There are many items in general we need to preserve that we are not doing right now.

whowutwut, 2017-09-14:
@bradbishop @anoo1 Maybe a good time to ask this again... With the new code update code coming, is anything like .ssh directory preserved when we flash the firmware?  We have a way to send over the ssh keys to the BMC, but after each FW flash, the authorized_keys file is wiped out.  Is that still the case? @chenglch FYI

williamspatrick, 2017-09-14:
Root's home directory will be preserved with the new code update. That code went in recently.

williamspatrick, 2017-09-14:
(We did that just for you! 😀)

whowutwut, 2017-09-16:
👍 Thanks @williamspatrick !  ( @gurevichmark , @chenglch, @zet809 FYI )

rfrandse, 2017-10-10:
#**STATUS**
10/10:
Pending Victor getting back to Michael - there’s a suspicion that it may be a problem with the XCAT tool and not a defect on openbmc end

whowutwut, 2017-10-12:
@rfrandse  I think this issue is different then the CQ issue I opened.  This issue is asking for a REST method for uploading a public key to the BMC....  Is there any ETA on this or even a statement that this request would be considered? 

The workaround that xCAT is doing to copy over a key to the BMC, is producing different success rates, that's what we are tracking with: https://github.com/xcat2/xcat-core/issues/4074

rfrandse, 2017-10-12:
@mtritz  since @whowutwut said this wasn't the same issue as SW403668
I went a head and bridged SW403668  and that's now connected to https://github.com/openbmc/openbmc/issues/2464 to track

closed 2019-10-27 by stale[bot]