openbmc/openbmc #1518: [Barreleye v1.99.4-34] Cannot boot host if FRU not filled correctly by BMC before poweron

yadron, 2017-04-25:
In
https://gerrit.openbmc-project.xyz/#/c/1701/ (and its source https://github.com/openbmc/openbmc/issues/665),
saying "In barellye this info was coming from the host" (it means FRU was coming, so it looks like Barreleye works differently from the other platforms and requires filling EEPROM/FRU by BMC).

It means that Barreleye will not boot correctly if eeprom/FRU not filled by BMC (at least with older PRONs, but I see this bug with latest PRON that I've compiled from sources too).

If Barreleye/host booting correctly (older v1.99.0-* OpenBMC vers), the process looks like (from journalctl):

```
Apr 25 17:31:44 s188 system_manager.py[756]: Running System State: HOST_BOOTING
Apr 25 17:31:44 s188 phosphor-host-state-manager[796]: Reached final state
Apr 25 17:31:44 s188 host_watchdog.exe[718]: Setting watchdog poll interval to: 30000
Apr 25 17:31:44 s188 host_watchdog.exe[718]: Starting watchdog with poll interval: 30000
Apr 25 17:31:49 s188 host_watchdog.exe[718]: Setting watchdog poll interval to: 30000
Apr 25 17:31:49 s188 host_watchdog.exe[718]: Stopping watchdog
Apr 25 17:31:49 s188 host_watchdog.exe[718]: Starting watchdog with poll interval: 30000
Apr 25 17:32:11 s188 ipmid[734]: Incomplete Fru file.. Size:[56]
Apr 25 17:32:11 s188 ipmid[734]: Populating FRU areas failed for:[1]
Apr 25 17:32:11 s188 inventory_items.py[727]: Caching: /org/openbmc/inventory/system/chassis/motherboard/membuf1
Apr 25 17:32:11 s188 inventory_items.py[727]: Caching: /org/openbmc/inventory/system/chassis/motherboard/membuf1
Apr 25 17:32:11 s188 inventory_items.py[727]: Caching: /org/openbmc/inventory/system/chassis/motherboard/membuf0
Apr 25 17:32:11 s188 inventory_items.py[727]: Caching: /org/openbmc/inventory/system/chassis/motherboard/membuf0
Apr 25 17:32:11 s188 inventory_items.py[727]: Caching: /org/openbmc/inventory/system/chassis/motherboard/membuf3
```

If cannot boot host (latest versions), then it looks like:

```
Apr 25 19:21:24 s188 system_manager.py[757]: Running System State: HOST_BOOTING
Apr 25 19:21:24 s188 host_watchdog.exe[693]: Setting watchdog poll interval to: 30000
Apr 25 19:21:24 s188 host_watchdog.exe[693]: Starting watchdog with poll interval: 30000
Apr 25 19:21:24 s188 kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Apr 25 19:21:24 s188 kernel[715]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
```
- and that's it. Nothing happens after that (even "obmcutil power" and "obmcutil state" says everything already booted).

Also, I'm interested why
`/usr/sbin/phosphor-read-eeprom --eeprom=/sys/bus/i2c/devices/0-0050/eeprom --fruid=1`
says "Failed to validate common header" (see phosphor-ipmi-fru/1.0-r1/git/writefrudata.cpp, line 565).

How to uploadl manually FRU to EEPROM (files "/tmp/ipmifru01,  ipmifru02,  ipmifru05,  ipmifru06,  ipmifru07,  ipmifru08,  ipmifru09,  ipmifru0a,  ipmifru0b,  ipmifru0c" to EEPROM)?
I want to fill it manually, because I think that it will be a workaround for starting HOST at last.

Let me explain again, that the problem for me that it looks like everything works
(see
```
root@barreleye:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```
),
but host boot frozen, because OpenPOWER CPU cannot say to CENTAUR "turn on memory power, everything ready", so host cannot start booting even BMC did all the job except of filling FRU.

The prev related issues of mine were:
https://github.com/openbmc/openbmc/issues/1287 - some architectural concerns here
https://github.com/openbmc/openbmc/issues/1262 - other similar issue regarding Palmetto
https://github.com/openbmc/openbmc/issues/832 - the issue that is being fixed right now (by GitHub:Geissonator) and possibly will help with 1287.
https://github.com/openbmc/openbmc/issues/1324 - the minor issue that is a part of https://github.com/openbmc/openbmc/issues/1287 and https://github.com/openbmc/openbmc/issues/1262 (like I've already written it in https://github.com/openbmc/openbmc/issues/1262 , but again for highlighting it)
https://github.com/openbmc/openbmc/issues/1358 - /usr/sbin/ipmid broken (error of loading libapphandler.so.0), fixed and closed.
https://github.com/openbmc/openbmc/issues/1452 - "obmcutil state" didn't work for v1.99.4-14 (and fixed with v1.99.4-15, but still cannot boot host)
https://github.com/openbmc/openbmc/issues/1454 - found the bug of "cannot boot host" (but looks like it helped before, because FRU was correct, and now I cannot boot anymore, because FRU incorrect again).

Who should fill FRU in newest versions of PNOR? Which service/part of PNOR?

@williamspatrick , you've said like I can contact hostboot team (open-power/hostboot) for fixing #1454 . Actually it looks like booting host frozen somewhere before "STEP 6.3" (it means that OCC itself or maybe other internal PNOR->OpenPOWER internal CPU code failed starting boot on very-very first stage even before CENTAUR initialized).

williamspatrick, 2017-04-25:
@yadron - Sorry, I'm really not understanding your concern here.

System vendors decide to put the BMC and/or system VPD in different locations.  On Barreleye, it was attached to the BMC so we never get it from the host code via IPMI.  There is nothing changed here except that we populate different inventory items now with newer code.

The log output looks exactly the same between the two except that, as you said, the host code isn't up far enough to send IPMI messages so you don't see any log events from that.

You are using very confusing terminology here.
```
but host boot frozen, because OpenPOWER CPU cannot say to CENTAUR "turn on memory power, everything ready", so host cannot start booting even BMC did all the job except of filling FRU.
```
```
Actually it looks like booting host frozen somewhere before "STEP 6.3" (it means that OCC itself or maybe other internal PNOR->OpenPOWER internal CPU code failed starting boot on very-very first stage even before CENTAUR initialized).
```

None of these are true statements from a system design.  The Centaur is the memory buffer, which isn't initialized until way past step 6.3.  The OCC is also not initialized until much later.  I think you mean SBE?  If you are not seeing any output from the serial console (never see "ISTEP 6.3" message) then there is a problem being identified by the SBE.  In any case, the host code / SBE teams would need to debug what is going on.  The BMC cannot even modify the SBE directly, so the only variable here is your PNOR image.

yadron, 2017-04-25:
Thank you for the info regarding SBE. Looks like SBE failed then (I've just looked to https://github.com/open-power/sbe/tree/master/src/boot). How to update SBE? Is it a part of overall PNOR that is in resulting file barreleye.pnor or should I update it separately?

I've read fast what is VPD and how it works for Barreleye.
It looks like it was a very old bug "https://github.com/open-power/hostboot/issues/29" that if one of CPUs not present, then cannot read VPD.
I have Barreleye with only one CPU slot used now out of two, so it looks for me like the same bug happens now for me.

How to upload VPD to PNOR/CVPD, so SBE of Barreleye (or Hostboot if this is not SBE) could read it from PNOR if it cannot read it directly from Centaur?

yadron, 2017-04-25:
@williamspatrick , also I see this line of code

`#if defined(CONFIG_CVPD_READ_FROM_HW) && defined(CONFIG_CVPD_READ_FROM_PNOR)
`

in /src/usr/fsi/fsipres.C of op-build, so do you think that rebuilding both BMC and PNOR with "CONFIG_CVPD_READ_FROM_PNOR=1 and CONFIG_CVPD_READ_FROM_HW=0" could help me?

yadron, 2017-04-28:
Seeing very strange things happens with inventory upload and FRU and other related.

Could you please help?

I see that if I have many memory DIIMS (just inserted many-many yesterday in ~Barreleye), the logic for filling FRU doesn't work anymore in ipmid (using older BMC version like 1.99.0.* that were stable before inserting more hardware).

I've just found related
https://github.com/openbmc/openbmc/issues/886
, and "parent" one
https://github.com/openbmc/openbmc/issues/681
, and it says

> FRU, IPMI, and OpenPower VPD inventory objects will require information from the MRW. This issue tracks getting that data into the MRW and then into a format that these objects can consume.

and

> All code required to generate YAML content for inband IPMI format
> 
> Fetch data from mrw
> -- Break down into parts (do processors, do memory, ...)
> Write script to generate yaml
> bitbake recipe
> Depends on Matt Spinler providing some required MRW data. Ratan to email Matt with details.

and

> write a perl script to fetch the data from MRW.
> 
>  Write a function which generate YAML file.
> 
>  Write a bitbake file.
> 

What is MRW?
Does this Pert script exists in OpenBMC? Can I use it manually?

- could you please share scripts or other info how to read MRW/etc manually (_and how to use this https://github.com/openbmc/phosphor-mrw-tools ?_), so how to read and program EEPROM manually, and how to prepare other data manually before starting IPMID and booting host?
What I already found is this cmd: "phosphor-read-eeprom --eeprom /sys/bus/i2c/devices/0-0050/eeprom --fruid 64" (source https://github.com/openbmc/ipmi-fru-parser/issues/19) and tried using it for seeing how reading eeprom and etc logic works.

What I'm going to do, I would like turning off "programming FRU" logic of ipmid and running it manually just before starting host (because ipmid itself buggy and sometimes does core-dump:

```
Apr 28 08:50:20 barreleye ipmid[748]: _append_to_dict: VPD Key = [Custom Field 8] : Type Code = [BINARY] : Len = [0] : Val = []
Apr 28 08:50:20 barreleye ipmid[748]: SUCCESS: Updated:[BOARD__13] successfully
Apr 28 08:50:19 barreleye systemd[1]: [[0;1;39morg.openbmc.HostServices.service: Failed with result 'core-dump'.[[0m
Apr 28 08:50:20 barreleye systemd[1]: org.openbmc.HostServices.service: Service hold-off time over, scheduling restart.
Apr 28 08:50:20 barreleye systemd[1]: Stopped Phosphor Inband IPMI.
Apr 28 08:50:20 barreleye systemd[1]: Starting Phosphor Inband IPMI...
Apr 28 08:50:20 barreleye systemd[1]: Started Phosphor Inband IPMI.
Apr 28 08:50:25 barreleye ipmid[888]: Failed to get /org/openbmc/settings/host0 busname: Device or resource busy
Apr 28 08:50:25 barreleye ipmid[888]: ERROR:[0xD4] handling NetFn:[0xA], Cmd:[0x12]
Apr 28 08:50:25 barreleye ipmid[888]: Failed to call the method: Input/output error
Apr 28 08:50:25 barreleye ipmid[888]: Dest: :1.1, Path: /org/openbmc/HostIpmi/1
```

during preparing FRU just before booting host).

@geissonator , also how MRW and GPIO num related to each other? For example, looking here
https://lists.ozlabs.org/pipermail/openbmc/2016-May/003162.html
it says "_+ * FIXME: Fetch GPIO NUM from MRW instead?_", what does it mean?
I've seen old ticket https://github.com/openbmc/openbmc/issues/527 where you had
```
  1.75122|  ModuleId   0x0a FSI::MOD_FSIDD_CHECKFORERRORS
  1.75123|  ReasonCode 0x0a11 FSI::RC_ERROR_IN_MAEB
```
and both GPIO problems and I've seen once this error too when inserting more memory DIMMs than few, so thinking if GPIO and many DIIMS and FRU and MRW related to each other.

Guys, if server works with many memory DIMMs and then you reboot it many times without problems and then you turn it off during night and then turn on again (7 hours later in the morning) without any changes, but it fails with errors like described above, how can you explain it?
Does it mean that filling FRU/upload_inventory/etc works randomly and no way to make it reliable, so I need to find a way (workaround) myself how to read it once during server works Ok, so then restoring it next time manually just before booting host (and both removing FRU/MRW/etc logic from ipmid)?

BTW, do anyone plan supporting POWER8 anymore or I maybe can forget about OpenBMC 2.0 regarding this Barreleye related problems? Asking this question, because for me it looks like Barreleye and similar platforms not supported by OpenBMC anymore starting from the beginning of March.

yadron, 2017-05-12:
This issue doesn't directly about host-boot. It is regarding "RC_ERROR_IN_MAEB" error. Possibly this error only happens if memory DIMM problem or riser problem, but still testing. I'll comment here after finishing.

gtmills, 2018-05-31:
@yadron Still a problem or can this be closed? A year+ old.

gtmills, 2018-07-05:
Closing. Nothing in over a year. Please reopen if still a problem. Barreleye is no longer a supported machine and this is a very old build.

closed 2018-07-05 by gtmills