openbmc/openbmc #1519: [openbmc#b5520b1] openpower-proc-control[1033]: filesystem error: directory iterator cannot open directory: No such file or directory [/sys/devices/platform/fsi-master/slave@00:00/hub@00/]

johnhcwang, 2017-04-26:
During the AC cycle looping test, sometimes we hit this failure and BMC fail to power on the system with power_policy 'ALWAYS_POWER_ON'.
```
Apr 25 10:03:20 zaius system_manager.py[856]: Running System State: HOST_POWERED_ON
Apr 25 10:03:20 zaius systemd[1]: Started Wait for Power0 to turn on.
Apr 25 10:03:20 zaius systemd[1]: Reached target Power0 On.
Apr 25 10:03:20 zaius systemd[1]: Reached target Power0 (On).
Apr 25 10:03:20 zaius timemanager[929]:  PGOOD has changed..
Apr 25 10:03:20 zaius timemanager[929]: Changing time settings is *deferred* now
Apr 25 10:03:20 zaius systemd[1]: Starting Scan FSI devices...
Apr 25 10:03:20 zaius flash_bios.exe[865]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Apr 25 10:03:20 zaius flash_bios.exe[865]: flasher: , BMC = 1, address = 0x0
Apr 25 10:03:20 zaius flash_bios.exe[865]: Setting up flash
Apr 25 10:03:20 zaius systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Apr 25 10:03:20 zaius systemd[1]: Starting Fix up Zaius host refclock...
Apr 25 10:03:21 zaius systemd-timesyncd[781]: Synchronized to time server 216.239.35.8:123 (time3.google.com).
Apr 25 10:03:21 zaius flash_bios.exe[865]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Apr 25 10:03:21 zaius flash_bios.exe[865]: flasher: , BMC = 1, address = 0x0
Apr 25 10:03:21 zaius flash_bios.exe[865]: Setting up flash
Apr 25 10:03:21 zaius systemd[1]: Started Fix up Zaius host refclock.
Apr 25 10:03:21 zaius systemd[1]: Started Scan FSI devices.
Apr 25 10:03:21 zaius systemd[1]: Starting Run VCS workaround on host0...
Apr 25 10:03:21 zaius openpower-proc-control[1033]: filesystem error: directory iterator cannot open directory: No such file or directory [/sys/devices/platform/fsi-master/slave@00:00/hub@00/]
Apr 25 10:03:21 zaius systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Apr 25 10:03:21 zaius systemd[1]: Failed to start Run VCS workaround on host0.
Apr 25 10:03:21 zaius systemd[1]: Dependency failed for Chassis0 (Power On).
Apr 25 10:03:21 zaius systemd[1]: Dependency failed for Start Host0.
Apr 25 10:03:21 zaius systemd[1]: obmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency'.
Apr 25 10:03:21 zaius systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Apr 25 10:03:21 zaius systemd[1]: Startup finished in 5.938s (kernel) + 1min 13.548s (userspace) = 1min 19.486s.
Apr 25 10:03:21 zaius systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
Apr 25 10:03:21 zaius systemd[1]: obmc-chassis-poweron@0.target: Triggering OnFailure= dependencies.
Apr 25 10:03:22 zaius systemd[1]: vcs_workaround@0.service: Unit entered failed state.
Apr 25 10:03:22 zaius systemd[1]: vcs_workaround@0.service: Failed with result 'exit-code'.
Apr 25 10:03:22 zaius systemd[1]: Created slice system-op\x2dpower\x2dstop.slice.
Apr 25 10:03:22 zaius systemd[1]: Created slice system-op\x2dwait\x2dpower\x2doff.slice.
Apr 25 10:03:22 zaius systemd[1]: Created slice system-obmc\x2dled\x2dgroup\x2dstop.slice.
Apr 25 10:03:22 zaius systemd[1]: Starting Assert power_on LED...
Apr 25 10:03:22 zaius systemd[1]: Created slice system-mboxd\x2dreload.slice.
Apr 25 10:03:22 zaius systemd[1]: Starting Wait for Power0 to turn off...
Apr 25 10:03:22 zaius systemd[1]: Started Stop Power0.
```

mine260309, 2017-04-26:
Regarding the version string `VERSION_ID="v1.99.4-53-ge850d91-dirty-john"`
John is using his revision: https://github.com/johnhcwang/openbmc/commit/e850d91bfe798539e84829515eaeed072377d5fd which is based on b5520b14b1d71f6365712017a3a2e647d4744965 on master branch.

He's using the kernel https://github.com/johnhcwang/linux/commit/93a1f08e75bf8393e0bd5214a7bb4f91905546c0 which disables ucd901690 as Witherspoon.

And he has local change to add `-john` in the version string.

So basically it is almost the same as OpenBMC master branch build.

spinler, 2017-04-26:
This is another case of the FSI scan not appearing to do anything.

cbostic, 2017-05-03:
Typically when FSI scan doesn't appear to do anything its due to the CFAM not being powered on.   'obmcutil poweron' should result in discovery of devices typically.  If that is being run then it would be worth trying a manual scan with:

echo 1 > /sys/bus/platform/devices/fsi-master/scan

I that doesn't help I can hop on a failing system and have a closer look.

johnhcwang, 2017-05-05:
We have a race condition between [op-refclock-fixup@.service](https://github.com/openbmc/openbmc/blob/v1.99.5/meta-openbmc-machines/meta-openpower/meta-ingrasys/meta-zaius/recipes-phosphor/chassis/refclock-fixup/op-refclock-fixup%40.service) and fsi-scan@.service.
Commit e1dcc23 mentions that the clock generator output configuration fixup should be started before trying to send any FSI transactions.

Pass over 300 test cycles with https://gerrit.openbmc-project.xyz/#/c/4074/

williamspatrick, 2017-05-05:
@johnhcwang please push fix to Gerrit and add @GHF for review.

mine260309, 2017-05-05:
The commit is already pushed to Gerrit: https://gerrit.openbmc-project.xyz/#/c/4074/
And I just added @GHF as reviewer.

closed 2017-05-09 by williamspatrick