openbmc/openbmc #1520: Barreleye- After code update unable to bring OS state to running state

prkatti1, 2017-04-26:
After code update unable to bring OS state to running state on Barreleye system.

Version details
*********************************************
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-60"
VERSION_ID="v1.99.4-60-g1fb2a9d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-60"
BUILD_ID="v1.99.4"

Linux barreleye 4.7.10-b6e95d08c1fd39409fe0d127007a4166024fc8e7 #1 Tue Apr 25 21:45:49 UTC 2017 armv5tejl GNU/Linux

PNOR details:
************************************************
IBM-barreleye-ibm-OP8_v1.15_2.4
        op-build-v1.15-96-gc612094
        buildroot-2017.02-4-gc28573e
        skiboot-5.4.3
        hostboot-695bd89
        linux-4.10.6-openpower1-p7669d59
        petitboot-v1.4.2-p1493d08
        barreleye-xml-253622f
        occ-301b535
        hostboot-binaries-5c78ef3
        capp-ucode-1bb7503

root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@barreleye:~# obmcutil poweron
root@barreleye:~# obmcutil power
pgood = 0
state = 1
pgood_timeout = 0
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

prkatti1, 2017-04-26:
It is going into Quiesce state during CT.

sivassrr, 2017-04-26:
@prkatti1 Cleaned up your first comment. You should remove ~ to have better readable format. 
Also your update not showing system reaching quiesce state. Is it really an issue?

prkatti1, 2017-04-26:
@sivassrr thanks for cleaning up. Yes it is really an issue. Currently it isin Quiesce state.
root@barreleye:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced

sivassrr, 2017-04-26:
Those who are looking at this issue, Please ping @prkatti1 for system details for your debug.

prkatti1, 2017-04-26:
[journalctl.txt](https://github.com/openbmc/openbmc/files/957894/journalctl.txt)
[dmesg.txt](https://github.com/openbmc/openbmc/files/957895/dmesg.txt)

yadron, 2017-04-26:
@prkatti1 , if I remember right, I've did workarounds for resolving similar issues:

1. Make larger timeout of watchdog everywhere (say 240000 instead of 30000). This could be done by edition of /lib/systemd/system/obmc-start-watchdog@.service using BMC console and some other 30000 I've edited in source code of host_watchdog_obj.c (possibly this one was not required).

2. I've edited obmc-power-chassis-on@%i.target also (removed second dep e.g. line 5 and 6 that are "Requires" and "After").

- only after these two fixes I can get 

```
root@barreleye:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```

, but still cannot boot host (that possibly non-regular problem, described here: https://github.com/openbmc/openbmc/issues/1518 ).

prkatti1, 2017-05-02:
Same problem observed on todays build

ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-78"
VERSION_ID="v1.99.4-78-g4fa6338"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-78"
BUILD_ID="v1.99.4"

geissonator, 2017-05-03:
The journal is just filled with these, could you make sure you have the latest test suite fixes in this area?  I thought we got a fix in to properly close our REST connections to prevent this:
```
14bb:cb4f:b04d:c3b6%3066252704', 443, 0, 3066252704L)>>(<socket at 0xb4c45730L fileno=1023>, ('::6059:c1b6:8e91:f402%1338680316', 36861, 0, 133)> failed with IOError
Apr 20 19:43:38 barreleye phosphor-gevent[757]: Traceback (most recent call last):
Apr 20 19:43:38 barreleye phosphor-gevent[757]:   File "/usr/lib/python2.7/site-packages/gevent/baseserver.py", line 140, in _do_read
Apr 20 19:43:38 barreleye phosphor-gevent[757]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 93, in do_read
Apr 20 19:43:38 barreleye phosphor-gevent[757]:     client_socket, address = self.socket.accept()
Apr 20 19:43:38 barreleye phosphor-gevent[757]: error: [Errno 24] Too many open files
Apr 20 19:43:38 barreleye phosphor-gevent[757]: <WSGIServer at 0xb65b3290L fileno=5 address=('::14bb:cb4f:b04d:c3b6%3066252704', 443, 0, 3066252704L)> failed with error
Apr 20 19:43:38 barreleye phosphor-gevent[757]: Traceback (most recent call last):
Apr 20 19:43:38 barreleye phosphor-gevent[757]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Apr 20 19:43:38 barreleye phosphor-gevent[757]:     result = self._run(*self.args, **self.kwargs)
Apr 20 19:43:38 barreleye phosphor-gevent[757]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Apr 20 19:43:38 barreleye phosphor-gevent[757]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Apr 20 19:43:38 barreleye phosphor-gevent[757]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Apr 20 19:43:38 barreleye phosphor-gevent[757]:     ciphers=ciphers)
Apr 20 19:43:38 barreleye phosphor-gevent[757]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 106, in __init__
Apr 20 19:43:38 barreleye phosphor-gevent[757]:     cert_reqs, ssl_version, ca_certs)
Apr 20 19:43:38 barreleye phosphor-gevent[757]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 31, in new_sslwrap
Apr 20 19:43:38 barreleye phosphor-gevent[757]:     context.load_cert_chain(certfile, keyfile)
Apr 20 19:43:38 barreleye phosphor-gevent[757]: IOError: [Errno 24] Too many open files
```
With all this data in the file, it's pretty much unreadable for me.  Please do a manual recreate and gather the appropriate logs?  I can't find a "quiesce" trace which I would expect.  Please also be sure to use a released PNOR image, not just the latest from master.

gtmills, 2018-05-31:
@prkatti1 This is a year+ old. Can this be closed? This is an old OpenBMC version.

prkatti1, 2018-06-01:
@gtmills yes, it can be closed.

closed 2018-06-01 by anoo1