openbmc/openbmc #1521: BMC Reset Reload TC fails due to Chassis is powered off

sivassrr, 2017-04-26:
```
Firmware Image Details:
- [x] BUILD_ID : https://openpower.xyz/job/openbmc-build/706
- [x] VERSION_ID : v1.99.4-60-g1fb2a9d
- [x] Kernel : Linux witherspoon 4.7.10-b6e95d08c1fd39409fe0d127007a4166024fc8e7 #1 Tue Apr 25 21:40:52 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.28
        op-build-v1.15-131-gda28b74-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-5.5.0
        hostboot-de49800
        linux-4.10.7-openpower1-pd408964
        petitboot-v1.4.2-p7985f01
        witherspoon-xml-07ef29d
        occ-e1889e5
        hostboot-binaries-d1b1312
        capp-ucode-1bb7503
        sbe-c9fea2e-p3dffe09
```
Test Case Ran: BMC Reset Reload when host is up and running. 
 Expected Result:  BMC Reset Reload happen without disturbing host power state. 
 Actual Result: BMC reset reload happened successfully but brought down the chassis power also as off. 

We are observing this frequently. 
[BMCResetReloadFailure.zip](https://github.com/openbmc/openbmc/files/958241/BMCResetReloadFailure.zip)

williamspatrick, 2017-04-26:
I think there is still work going on for this support.

williamspatrick, 2017-04-26:
@geissonator : Feel free to close this and reference your stories to add this support.

sivassrr, 2017-04-26:
That helps @williamspatrick

sivassrr, 2017-04-27:
Oops, I found the equivalent story https://github.com/openbmc/openbmc/issues/1274.
So closing as dup.

sivassrr, 2017-04-27:
1274 talks about Host. Need issue for Chassis.

rfrandse, 2017-05-02:
Since this was reopened marking as bug

geissonator, 2017-05-15:
All code for this is in, and can be verified on barreleye (and maybe Zaius).  But on witherspoon, we have #1322.  Closing this one.

closed 2017-05-15 by geissonator