openbmc/openbmc #1523: i2cfsi device issue eventually losing connection to device

janssens2, 2017-04-26:
Seeing an issue running on a zaius machine (zaius1) where running over a fsp2 connection for fsi is successful.  When using the i2cfsi device on the bmc the connection appears to degrade over time.

Example of degradation is when fsp2 connected I can detect 8 dimms on each processor.  When running the same process on i2cfsi device I see all 8 dimms on proc 0, but on proc 1 a mixture report in leaving only 4 being discovered.  On a subsequent read a subset of those 4 found are then detected.

Noticed the spd reads appear to have some bad data in them.

```
0000000: 2311 0c01 8521 0008 0060 0003 090b 8000  #....!...`.   0000000: 2311 0c01 8521 0008 0060 0003 090b 8000  #....!...`.
0000010: 0000 070d f80f 0000 6e6e 6e11 006e f00a  ........nnn   0000010: 0000 070d f80f 0000 6e6e 6e11 006e f00a  ........nnn
0000020: 2008 0005 00a8 1b28 2800 7800 143c 0000   ......((.x   0000020: 2008 0005 00a8 1b28 2800 7800 143c 0000   ......((.x
0000030: 0000 0000 0000 0000 0000 0000 0b2c 0b2c  ...........   0000030: 0000 0000 0000 0000 0000 0000 0b2c 0b2c  ...........
0000040: 0b36 2c36 0b2c 0c2c 0c2c 0c2c 0c2c 0000  .6,6.,.,.,.   0000040: 0b36 2c36 0b2c 0c2c 0c2c 0c2c 0c2c 0000  .6,6.,.,.,.
0000050: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000050: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000060: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000060: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000070: 0000 0000 0000 9cb5 0000 0000 e7d6 6a66  ...........   0000070: 0000 0000 0000 9cb5 0000 0000 e7d6 6a66  ...........
0000080: 0a11 ff15 0086 32a0 0110 0000 0000 0000  ......2....   0000080: 0a11 ff15 0086 32a0 0110 0000 0000 0000  ......2....
0000090: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000090: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
00000a0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00000a0: 0000 0000 0000 0000 0000 0000 07ff ffff  ...........
00000b0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00000b0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00000c0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00000c0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00000d0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00000d0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00000e0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00000e0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00000f0: 0000 0000 0000 0000 0000 0000 0000 d3ca  ........... | 00000f0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
0000100: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000100: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000110: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000110: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000120: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000120: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000130: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000130: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000140: 0198 4b16 0860 8ff2 084b 4632 3730 3030  ..K..`...KF   0000140: 0198 4b16 0860 8ff2 084b 4632 3730 3030  ..K..`...KF
0000150: 3039 332d 3030 3220 2020 2020 2001 80ce  093-002       0000150: 3039 332d 3030 3220 2020 2020 2001 80ce  093-002
0000160: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000160: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000170: 0000 0000 0000 0000 0000 0000 0000 0000  ...........   0000170: 0000 0000 0000 0000 0000 0000 0000 0000  ...........
0000180: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 0000180: 0000 0000 0000 0000 0000 0000 7fff ffff  ...........
0000190: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 0000190: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00001a0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00001a0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00001b0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00001b0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00001c0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00001c0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00001d0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00001d0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00001e0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00001e0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
00001f0: 0000 0000 0000 0000 0000 0000 0000 0000  ........... | 00001f0: ffff ffff ffff ffff ffff ffff ffff ffff  ...........
```

This shows a dump of the spd (left is fsp2 based dump) (right is i2cfsi based dump)  This is read in 256 byte chunks.  Notice the 7FFF....    These started first with a 7FFFFFFFFFFFFFFFFFFFFFFF in the read.  Above is the read after noticing the first set of 7FF and generates many more FFFFF's than expected.

After a period of time or reads (not sure it makes a difference)  we eventually cannot talk to proc 1 and the dimm's on proc 0 go away too.  This appears by running the iicmaster --scan on the bmc too.

Attaching dmesg output from initial crodetcnfg and then a followup crodetcnfg where dimm's disappeared.

[zaius1.dmesg.2.txt](https://github.com/openbmc/openbmc/files/959774/zaius1.dmesg.2.txt)
[zaius1.dmesg.1.txt](https://github.com/openbmc/openbmc/files/959775/zaius1.dmesg.1.txt)

cbostic, 2017-10-16:
@janssens2 Is this still an issue?

janssens2, 2017-10-16:
@cbostic  I have not encountered this issue recently.  I think it may have been resolved in some other commits that Eddie made earlier.  I am not able to recreate this at the moment.

closed 2019-04-27 by stale[bot]