openbmc/openbmc #1525: [Palmetto/Witherspoon] Reading attribute using REST is failing with latest build

swe12345, 2017-04-27:
Reading attribute using REST is failing with latest build 709/v1.99.4-69-gb2ca901. Due to this all automation test cases are failing. So it need to be fixed on priority.

This issue is seen with all platform.
```
bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/host0/attr/CurrentHostState
{
  "data": {
    "description": "Expecting content type 'application/json', got ''"
  },
  "message": "415 Unsupported Media Type",
  "status": "error"
}bash-4.1$
```

BMC Fw:
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-69"
VERSION_ID="v1.99.4-69-gb2ca901"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-69"
BUILD_ID="v1.99.4"
root@palmetto:~# cat /etc/timestamp
20170427032800
```

prkatti1, 2017-04-27:
Same issue with barreleye

ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-69"
VERSION_ID="v1.99.4-69-gb2ca901"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-69"
BUILD_ID="v1.99.4"

dkodihal, 2017-04-27:
Fix up for review at https://gerrit.openbmc-project.xyz/#/c/3939/1.

sivassrr, 2017-04-27:
@prkatti1 We are seeing this even in Barreleye as well, right. Could you confirm. That helps.

closed 2017-04-27 by williamspatrick