openbmc/openbmc #1529: Witherspoon: pgood poll timeout

amboar, 2017-04-28:
Nico Fajardo reported he had a Witherspoon system failing to boot the host after several successful attemps.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.4"
VERSION_ID="ibm-v1.99.4-0-r0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.4"
BUILD_ID="ibm-v1.99.4-0-r0"
```
The [journal output](https://github.com/openbmc/openbmc/files/963641/pgood-timeout.journal.txt) doesn't show any obvious failures, it just seems that we fail to receive the pgood signal from the UCD90160. Eagle-eyed readers might note that the pgood timeout value extended to a minute in the last boot attempt - I experimented briefly to see whether doubling the timeout value would get us into a good state. It didn't.

Further, the UCD device state doesn't appear to have been corrupted and so should be in working order:

```
root@witherspoon:~# ./dumpucd.sh 
./dumpucd.sh: line 18: echo: write error: No such device
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 1
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 2
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 3
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 4
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 5
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 6
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 7
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 8
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 9
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 10
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 11
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 12
32: 0x16 0xfe 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 13
32: 0x1a 0xda 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 14
32: 0x1a 0xda 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
Finished with 15
```

Finally, Nico noted that the problem could often be resolved by power-cycling the system at the PDU.

paulgrobertson, 2017-05-05:
suspect you have a reg that is not coming on.   on next occurrence, run vrm-control.sh to see which rails fail to power on.

NicoFajardo, 2017-05-12:
Updated to latest BMC code:
root@witherspoon:/tmp# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.5"
VERSION_ID="ibm-v1.99.5-0-r0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.5"
BUILD_ID="ibm-v1.99.5-0-r0"

Saw the issue again. None of the regulators were coming online and the Chassis.PowerState was never reaching "On" state. After digging through the journalctl debug log:

> May 03 15:27:08 witherspoon phosphor-host-state-manager[861]: Initial Host State will be Off
May 03 15:27:08 witherspoon systemd[1]: Started Phosphor Host State Manager.
May 03 15:27:08 witherspoon phosphor-bmc-state-manager[862]: Setting the BMCState field
May 03 15:27:09 witherspoon host_checkstop.exe[858]: GPIO Lookup:  CHECKSTOP = 354,falling
May 03 15:27:08 witherspoon phosphor-bmc-state-manager[862]: Setting the BMCState field
May 03 15:27:08 witherspoon systemd[1]: Started Phosphor BMC State Manager.
May 03 15:27:08 witherspoon phosphor-chassis-state-manager[857]: Initial Chassis State will be Off

...

> May 12 00:34:20 witherspoon systemd[1]: Started Assert power_on LED.
May 12 00:34:20 witherspoon vrm-control.sh[969]: rail        set        read        current
May 12 00:34:20 witherspoon vrm-control.sh[969]: -------        ------- ------- -------
May 12 00:34:21 witherspoon vrm-control.sh[969]: vdna        0.898V        0.004V        0.000A
May 12 00:34:21 witherspoon vrm-control.sh[969]: vdnb        0.898V        0.004V        0.000A
May 12 00:34:21 witherspoon systemd[1]: Started Apply voltage overrides to VRMs.
May 12 00:34:21 witherspoon systemd[1]: Starting Enable the AVS bus on VRMs...
May 12 00:34:21 witherspoon systemd[1]: Started Enable the AVS bus on VRMs.
May 12 00:34:21 witherspoon systemd[1]: Starting Disable VCS before Witherspoon power on...
May 12 00:34:21 witherspoon systemd[1]: Started Disable VCS before Witherspoon power on.
May 12 00:34:21 witherspoon systemd[1]: Reached target Power0 On (Pre).
May 12 00:34:22 witherspoon systemd[1]: Starting Start Power0...
May 12 00:34:22 witherspoon power_control.exe[860]: PowerControl: setting power up SOFTWARE_PGOOD to 1
May 12 00:34:22 witherspoon power_control.exe[860]: PowerControl: setting power up BMC_POWER_UP to 1
May 12 00:34:22 witherspoon system_manager.py[813]: Running System State: HOST_POWERING_ON
May 12 00:34:22 witherspoon timemanager[894]:  PGOOD has changed..
May 12 00:34:22 witherspoon systemd[1]: Started Start Power0.
May 12 00:34:34 witherspoon power_control.exe[860]: ERROR PowerControl: Pgood poll timeout
May 12 00:34:34 witherspoon timemanager[894]:  PGOOD has changed..
May 12 00:34:48 witherspoon systemd[1]: op-wait-power-on@0.service: Start operation timed out. Terminating.
May 12 00:34:48 witherspoon systemd[1]: op-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM
May 12 00:34:48 witherspoon dropbear[927]: Exit (root): Disconnect received
May 12 00:34:48 witherspoon systemd[1]: Failed to start Wait for Power0 to turn on.
May 12 00:34:48 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).

amboar, 2018-05-23:
Haven't heard further reports. There has been significant work on Witherspoon in the interim, so closing as no longer relevant.

closed 2018-05-23 by amboar