openbmc/openbmc #1531: Need a method to default cooling properties

bjwyman, 2017-04-28:
Issue #370 introduces an application that can be used to read a GPIO to determine if the WaterCooled property should be set to true or false, if the AirCooled property should be set to true or false. There is an option of having it set one or the other to true, but there should probably be a method to default AirCooled to true, in case a system type does not need the application for reading a GPIO.

@bradbishop was thinking that phosphor-inventory-manager would be a good way to default those properties, possibly through some configuration (YAML?) file.

msbarth, 2019-03-28:
Resolved by https://gerrit.openbmc-project.xyz/19781

At this time, testing is still needed to be done that persisted property values are not overwritten by a configured setProperty within pim's startup events. i.e.) Only set the property value if no default value or persisted value exists

rfrandse, 2019-03-28:
<https://gerrit.openbmc-project.xyz/c/openbmc/meta-ibm/+/19781> wspoon: Default cooling type to air only
Resolves  #1531 Need a method to default cooling properties

gtmills, 2019-11-16:
@msbarth Still a problem ?
Still planning on getting https://gerrit.openbmc-project.xyz/c/openbmc/meta-ibm/+/19781 in?

closed 2020-06-16 by stale[bot]