openbmc/openbmc #1533: BuildDate not getting updated under motherboard fru record

ploetzma, 2017-04-28:
Hostboot has started to read the Mfg. Date / Time VPD data for the backplane and add it to the record to be sent down to the BMC to be displayed with fru inventory.

When adding this data to a Witherspoon system, I do not see the data make its way down to the BMC. I can alter the Board Product name in VPD and that change shows up. I also applied this change to a Boston system (SuperMicro ATEN stack) and the Build date is showing up. Is there a different way to send an IPMI packet for openBMC?

What I am seeing on a Witherspoon BMC (altering the PrettyName to show we are correctly sending data):
```    "/xyz/openbmc_project/inventory/system/chassis/motherboard": {
      "BuildDate": "",
      "Manufacturer": "0000000000000000",
      "Model": "",
      "PartNumber": "0000000",
      "Present": 0,
      "PrettyName": "SZSTEM PLANAR   ",
      "SerialNumber": "000000000000"
```
io_data 4-6 is the hex value of the build date:

```
 28.29947|IPMI|MATT TRACE after insert io_data[0] is 0x1
 28.29949|IPMI|MATT TRACE after insert io_data[1] is 0xA
 28.29950|IPMI|MATT TRACE after insert io_data[2] is 0x0
 28.29951|IPMI|MATT TRACE after insert io_data[3] is 0xDF
 28.29953|IPMI|MATT TRACE after insert io_data[4] is 0x79
 28.29954|IPMI|MATT TRACE after insert io_data[5] is 0xA4
 28.29956|IPMI|MATT TRACE after insert io_data[6] is 0xD0

```

ratagupt, 2017-05-10:
Need journald logs to debug..
or get me the command that how do you change the build date.

ratagupt, 2017-05-29:
How to reject the issue saying need more info?

williamspatrick, 2017-05-29:
Just contact Matt directly for more info.

ploetzma, 2017-05-30:
Loaded w37 with a pnor that had VPD data injected with a build date for the backplane

Hostboot traces where we process the vpd data and send down to BMC via ipmi (not much here, but just for context):
``` 15.45566|IPMI|IPMIFRUINV::setData - Collecting all IPMI FRU Inventory Targets with fruId: [00000003]^M
 15.45567|IPMI|fru: I>sending write_fru_data fru size 56 offset 0^M
 15.45568|IPMI|rp: queuing sync 28:12^M
 15.45571|IPMI|backplaneIpmiFruInv::buildChassisInfoArea -  Using NEW OSYS RECORD FOR Chassis VPD Data^M
 15.45579|IPMI|dd: I>write ok 28:12 seq 7 len 3b^M
 15.45587|IPMI|IPMIFRUINV::setData - Sending IPMI FRU Inventory Data for target with fruId: [00000003] and size [00000001]^M
 15.45588|IPMI|fru: >>writeData(deviceId 0x3 size 128 offset 0)^M
 15.45590|IPMI|fru: <<writeData^M
```

inventory.sh ran from lcb325 after w37 was at ubuntu:
```    "/xyz/openbmc_project/inventory/system/chassis/motherboard": {
      "BuildDate": "",
      "Manufacturer": "0000000000000000",
      "Model": "",
      "PartNumber": "0000000",
      "Present": 0,
      "PrettyName": "SYSTEM PLANAR   ",
      "SerialNumber": "000000000000"
    },
```

Journal logs collected, here in gist:
https://gist.github.com/ploetzma/b49e0ad87732dd502cae30f05c78b503

ratagupt, 2017-05-31:
I don't see that host boot is sending the buildate to the openbmc through IPMI,
below are the IPMI traces taken from the journal logs.

ploetzma, 2017-05-31:
Spoke with Ratan last night:
1,) Build date is missing from the above logs (for openBmc). It appears Ratan found an ifdef that was not defined, causing it to skip printing the build date ipmid logs. He tried to turn it on, on w37.
2,) I printed out the whole IPMI message we are sending:
``` 20.75718|IPMI|MAP for RATAN - FRU Data:
~[0x0000] 01000106 000000f8 010502d0 30303030    *............0000*
~[0x0010] 30303030 30303030 30303030 d0303030    *000000000000.000*
~[0x0020] 30303030 30303030 30303030 3000c197    *0000000000000...*
~[0x0030] 010a0072 a1a1d030 30303030 30303030    *...r...000000000*
~[0x0040] 30303030 303030d0 53595354 454d2050    *0000000.SYSTEM P*
~[0x0050] 4c414e41 52202020 d0303030 30303030    *LANAR   .0000000*
~[0x0060] 30303030 30000000 00d03030 30303030    *00000.....000000*
~[0x0070] 30000000 00000000 000000c1 0000008d    *0...............*
 20.75720|IPMI|fru: <<writeData```

And the specific build date data I am computing:
``` 20.64987|IPMI|MATT TRACE - o_mfgDate is 10592626
 20.64988|IPMI|MATT TRACE - leastSig is 0x072
 20.64989|IPMI|MATT TRACE - middleSig is 0x0a1
 20.64991|IPMI|MATT TRACE - mostSig is 0x0a1
 20.64992|IPMI|MATT TRACE - io_data[0] is 0x01
 20.64994|IPMI|MATT TRACE - io_data[1] is 0x00
 20.64995|IPMI|MATT TRACE - io_data[2] is 0x00
 20.64996|IPMI|MATT TRACE - io_data[3] is 0x072
 20.64997|IPMI|MATT TRACE - io_data[4] is 0x0a1
 20.64999|IPMI|MATT TRACE - io_data[5] is 0x0a1 ```

Hostboot is sending down 72A1A1 as build date data (least sig first).

That data is found in the IPMI message from bytes 0x0034 to 0x0036.

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6166> Support Build date in IPMI Board FRU format
Resolves  #1533 BuildDate not getting updated under motherboard fru record

closed 2017-08-23 by williamspatrick