openbmc/openbmc #1534: OpenFSI: firmware assisted IPL not working  (SBE FIFO)

yadlapati, 2017-04-28:
Getting following error on Witherspoon and Zaius:
$ istep
CNFG FILE GLOBAL_DEBUG: L.H
Running Step 'startipl(0)' on Cage k0  : Completed

Running Step 'proc_sbe_attr_setup(2)' on Cage k0 Response header invalid: wrong magic code value [0000] l_totallengthInWords [00001403]ERROR: (Cro9PuChip::sbei_sbefifo_executeCronusIstep): sbei_sbefifo_executeIstep call failed rc= 37755904
ERROR: (ECMD): istep - Problems running iSteps
=============== EXTENDED ERROR MSG : Cronus ===============
ERROR: (BusMasterFsi::scomout): Problem calling interface: rc = 37755904(SERVER_INVALID_SBEFIFO_DEVICE)

root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-74"
VERSION_ID="v1.99.4-74-gdd36a3e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-74"
BUILD_ID="v1.99.4"

Eddie built a driver few days ago from his sandbox and that driver worked with out any issue. Build from Jenkins with all changes integrated is not working

eddiejames, 2017-05-01:
Ah. Turns out SBEFIFO is just not enabled in the kernel config... will fix.

gtmills, 2018-05-31:
@yadlapati No update in a year. Still a problem? Can this be closed?

closed 2019-04-27 by stale[bot]