openbmc/openbmc #1537: Sprint 17.05.05 - Committed

rfrandse, 2017-05-01:


dkodihal, 2017-05-04:
Squad 1 sprint plan

**Deepak**
**17.04.19**
- #1486 phosphor-settingsd : move to new d-bus interfaces 
- #1487 phosphor-settingsd : implement d-bus interfaces

**17.05.03**
- #1557 Design proposal to preserve and restore d-bus objects

**Ratan**
**17.05.03**
- #1479 mboxd: Handle write back to written sections
- <Network refactor story - priority is we need UT this sprint>

**Jayanth**
**17.04.05**
- #1378 Define Tooling infrastructre for Dump Parsing

**Tom**
**17.05.03**
- #1515 SEL: Implement the mandatory SEL commands
- #1516 SEL: Write infrastructure code to convert a Error Log Entry to SEL

**Dhruv**
**17.05.03**
- #1558 Non-sensor refactor : MRW parser and yaml design 

**Vishwa**
**17.05.03**
- #1288 led: Lamp Test

**Devender**
**17.04.19**
- #1469 Error-log : enable open-powerd-bus and application local errors

**17.05.03**
<IPMI story>

bradbishop, 2017-05-04:
-------------------------------------------------------------------------------------------------
***Squad 2 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Matt:spinler -** 
**17.01.11:**
- #629 (integrate device tree into kernel) - waiting

**17.03.22:**
- #1380 (Create fan group definition YAML) - up for review
    - 4/27 - almost merged
    - 05/01 - one commit left
    - 05/02 - one commit left

**17.04.05:**
- #1445 (Create fan control app and set initial fan speeds ) - up for review
    - 4/26 - working thru code review comments
    - 4/27 - almost merged
    - 05/01 - almost ready
    - 05/02 - two commits merged, getting closer

**17.04.19:**
- #1346 - tach feedback func state monitoring
    - 4/21 good progress on this yesterday
    - 4/24 - more good progress
    - 4/28 - made some progress - timer utility class
    - 05/01 - did a mako/python, make recipe today
    - 05/02 - done

-------------------------------------------------------------------------------------------------
**Matt B.:msbarth -**
**17.03.22:** 
- #1343 (Sensor groups)
    - 4/21 addressing review feedback
    - 4/24, 4/25 - got thru half
    - 4/26 - got thru the rest, waiting for brads review

**17.04.19:**
- #1493 (sensor parser)
    - 4/21 worked this.
    - 4/24, 4/25 still working
    - 4/26 still working it
    - 4/27 moved to brad
    - 4/29 good progress

- #1186 (system shutdown) - actual not committed
    - 4/27 - moved to backlog

- #944 (degraded mode)
    - 4/27 - going to work this
    - 4/29 - working on yaml and parser 
    - 05/01 - working on code
    - 05/02 - still working, getting close but not ready today.
-------------------------------------------------------------------------------------------------
**Brad:bradbishop -**
**17.03.08:**
- #1257 (sbefifo driver base) - out for review

**17.03.22:**
- #1303 (inventory Monitoring - Determine design and yaml input layout) 
     - 4/26 started looking
     - 05/01 work in progress

-------------------------------------------------------------------------------------------------
**Chris -**
**17.03.08:**
- #1219 (OpenFSI targeted error handling for mulitple hub links)
   - 4/24 - in 4.10 submit

**17.03.22:**
- #1222 ( OpenFSI: interrupt support for hub links.)
   - 4/24 - up for review

**17.04.19**
 - need a story
 - 4/25 working on an FSP defect
 - #1536 (document raw/scan files)
    - 05/02 - mostly done

------------------------------------------------------------------------------------------------
**Eddie:eddiejames -**
**16.12.28:**
- #624 (initial occ intf refactor) - waiting on community
- #625 (p9 occ data) - waiting on community

**17.02.08:**
- #1004 (I2C migration to official interfaces) - need a review

**17.03.22:**
- #1258 (sbefifo in kernel api) - need a review

- #1348 (OCC: force hwmon attribute numbering to conform to dts sensor definitions) 

**17.04.19:**
- #1491 (new endian requirements from openfsi in 4.10)
   - 4/25 - got it working
   - 4/26 - pushed commits, work for both 4.7 and 4.10
   - 4/27 - working thru code review
   - 4/28 - ready for review
   - 05/02 - ready for patrick
- #1492 (4.10 kernel integration)
   - 4/24 - working on it.  
   - 4/26 - working on i2c driver
   - 4/27 - still working on i2c driver
   - 05/02 - possibly switching to new i2c driver

- #1526 - pdbg updates
   - 4/28 - started yesterday, target today possibly
    - 05/02 - stop working, JK did this.
-------------------------------------------------------------------------------------------------
**Brandon Wyman -**
**17.03.22:**
- #370 ( Fan control - Air cool vs Water cool policies.)
    - 04/24 - trouble building witherspoon
    - 04/25 - plan on testing today
    - 04/26 - did some testing
    - 04/27 - incorporated review feedback
    - 04/28 - incorporated, need review, started looking into recipe updates
    - 05/01 - worked on recipes, addressed review comments, did reviews
    - 05/02 - recipes up.  incorporate feedback today.

-------------------------------------------------------------------------------------------------
**Dinesh Chinari  -**
- #1315 (Ensure uboot fw_env update script uses new inventory object) 
      - 04/24 - almost there, getting an exception
      - 04/25 - tested successfully on barreleye, will put up for review today
      - 04/27 - up for review, working thru feedback
      - 04/28 - still working thru feedback
      - 05/02 - up for review, working thru feedback

- #1322 (pgood is lost on BMC reboot on witherspoon)
      - 04/24 - characterizing the problem further - might be multiple bugs

gtmills, 2017-05-04:
# Squad 3 -

 **Adriana:anoo1 -**

**17.03.08:**

- #1297 (eSEL Missing AdditionalData [ Witherspoon ])

  * 05/04: 

- #1344 (Block State Transitions during Image Activation)

  * 05/04:  

**17.04.05:**

- #1278 Create ubi volumes when requested activation

  * 05/04: 

**17.04.19:**

- #1478 mboxd: Handle multiple mount types
  * 05/04:

closed 2017-05-25 by rfrandse