openbmc/openbmc #1545: Walking inventory paths doesn't always give FieldReplaceable : True

causten, 2017-05-02:
"/xyz/openbmc_project/inventory/system/chassis/motherboard"

  "/xyz/openbmc_project/inventory/system/chassis/motherboard": {
      "BuildDate": "", 
      "Manufacturer": "0000000000000000", 
      "Model": "", 
      "PartNumber": "0000000", 
      "Present": 0, 
      "PrettyName": "SYSTEM PLANAR   ", 
      "SerialNumber": "000000000000"
    }

The motherboard is a fru right?  The property just doesn't exist.  something in capturing the data seems off since the 'presence' property is also set to 0.

"/xyz/openbmc_project/inventory/system/chassis/motherboard/fan0"
- Fans are certainly replaceable yet the property does not exist there either.

--- Steps to recreate ---
Version: v1.99.4-82-g874f12e
Powered on the hardware. Issue this command...
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://x.x.x.x/xyz/openbmc_project/enumerate

gkeishin, 2017-05-03:
I think this is the dupe of https://github.com/openbmc/openbmc/issues/1216

causten, 2017-05-03:
It is not exactly a dup but getting a fix for 1216 may actually fix this issue too.

gtmills, 2018-05-23:
``` curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET  https://${BMC_IP}/xyz/openbmc_project/inventory/system/chassis/motherboard
{
  "data": {
    "BuildDate": "1996-01-01 - 00:00:00", 
    "Cached": 0, 
    "FieldReplaceable": 0, 
    "Manufacturer": "IBM WS", 
    "Model": "", 
    "PartNumber": "00VK672         ", 
    "Present": 1, 
    "PrettyName": "FRU DESCRIPTION", 
    "SerialNumber": "YL30UF77V007    "
  }, 
  "message": "200 OK", 
  "status": "ok"
}bash-4.1$curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET  https://${BMC_IP}/xyz/openbmc_project/inventory/system/chassis/motherboard/fan0
{
  "data": {
    "Functional": 1, 
    "MeetsMinimumShipLevel": 1, 
    "Present": 1, 
    "PrettyName": "fan0"
  }, 
```
I think this is still a problem.

closed 2019-04-27 by stale[bot]