openbmc/openbmc #1546: DIMM pretty name isnt pretty

causten, 2017-05-02:
See the pretty name... For a DIMM it is "0x0c".  You now what would be a great pretty name?  "16GB".  
    
  "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm4": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Functional": 1, 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G6B1   ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x156d5d85", 
      "Version": "0x31"
    }, 

--- steps to recreate ---
Version: v1.99.4-82-g874f12e
Powered on the hardware.  Issue this command... 
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET   https://x.x.x.x/xyz/openbmc_project/enumerate

gkeishin, 2017-05-03:
This one is also have a CQ FW665454  tracking it https://github.com/openbmc/openbmc/issues/1169

AlexanderAmelkin, 2018-04-10:
I believe this has nothing to do with OpenBMC and must be fixed in open-power firmware instead.
Please see https://github.com/open-power/hostboot/issues/132

AlexanderAmelkin, 2018-04-10:
IMO, respective issues must be filed with firmwares for other architectures as applicable.

gtmills, 2018-05-23:
Still see this 
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET  https://${BMC_IP}/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm0
{
  "data": {
    "BuildDate": "", 
    "Cached": 0, 
    "FieldReplaceable": 1, 
    "Functional": 1, 
    "Manufacturer": "0x2c80", 
    "Model": "36ASF4G72PZ-2G6D1   ", 
    "PartNumber": "", 
    "Present": 1, 
    "PrettyName": "0x0c", 
    "SerialNumber": "0x1434d95b", 
    "Version": "0x31"
  },
```

spinler, 2018-05-23:
I agree with @AlexanderAmelkin that if it's going to be fixed it should be fixed on the hostboot side.  Closing.

closed 2018-05-23 by spinler

AlexanderAmelkin, 2018-09-21:
JFYI, it has been fixed in https://github.com/open-power/hostboot/pull/147