openbmc/openbmc #1548: How to set boot media and boot mode?

xuweibj, 2017-05-03:
Hi, I want to set boot media and boot mode that should be used on the next boot, but I do not know what the interface. 
Is this function ready? Or what is the plan and design?

williamspatrick, 2017-05-03:
The function is there to send those up to the host but the REST API for "settings" is currently being reworked.  Do you want the old API now or do you want to wait for the final API?

williamspatrick, 2017-05-03:
@dkodihal - assigned you because you are working on the settings refactor.

xuweibj, 2017-05-04:
If possible, I hope I can get the design of final API now. But if the design is not ready now, may I know when it will be ready?

dkodihal, 2017-05-12:
Could you clarify what you expect with each of these?

boot_media - does this identify the source of the host image?
boot_mode - what is this for?

xuweibj, 2017-05-15:
Hi @dkodihal , 

If use ipmitool command, output will be like this:

```
# ipmitool chassis bootdev
bootdev <device> [clear-cmos=yes|no]
bootdev <device> [options=help,...]
  none  : Do not change boot device order
  pxe   : Force PXE boot
  disk  : Force boot from default Hard-drive
  safe  : Force boot from default Hard-drive, request Safe Mode
  diag  : Force boot from Diagnostic Partition
  cdrom : Force boot from CD/DVD
  bios  : Force boot into BIOS Setup
  floppy: Force boot from Floppy/primary removable media 
```

xuweibj, 2017-06-07:
Hi @dkodihal , may I know the interface now?

dkodihal, 2017-06-07:
@xuweibj : Here are the intefaces - https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Control/Boot. Let me know if you have question about these.

dkodihal, 2017-06-07:
The consumption of this interface by the IPMI code is still to be completed. I intend to take that up in our next sprint (mid June).

xuweibj, 2017-06-08:
@dkodihal Thank you for your reply. Will the request and response like below?
To get boot source:
```
# curl -k -c cjar -X GET https://bmc/xyz/openbmc_project/control/boot/bootsource 
{"status": "ok", "message": "200 OK", "data": {“BootSource”: “xyz.openbmc_project.Control.Boot.Sources.Disk”}}
```
To set boot source:
```
# curl -k -c cjar -X PUT -H "Content-Type: application/json"  https://bmc/xyz/openbmc_project/control/boot/bootsource/attr/Sources/ -d '{"data": "xyz.openbmc_project.Control.Boot.Sources.Network”}’
{"status": "ok", "message": "200 OK"}
```

If there is anything I understand wrong, please tell me.

xuweibj, 2017-06-20:
Hi @dkodihal , is there any progress about API code? And could you help to give comments to my last question?

dkodihal, 2017-06-22:
@xuweibj the code should be done in a week and merged in two. I'm using issue #1489 to track that.

The request and response from your previous comment seem right.

xuweibj, 2017-06-23:
@dkodihal , that sounds good. Thank you for your reply.

xuweibj, 2017-08-01:
Got interface, so close this issue.

closed 2017-08-01 by xuweibj