openbmc/openbmc #1549: BMC went into bad state and unable to ssh

prkatti1, 2017-05-03:
witherspoon system went into bad state when running CT and inband ipmi was enabled while running.
During this testing BMC went into bad state where IP was pingable but we were unable ssh and even we were unable to access serial port console. To bring system into normal state we did a dediprog to get the latest 0penBMC code on flash. We dont have any logs as system was un accessible.

prkatti1, 2017-05-03:
Version Info-
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-82"
VERSION_ID="v1.99.4-82-g874f12e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-82"
BUILD_ID="v1.99.4"

gkeishin, 2017-05-03:
Dupe of https://github.com/openbmc/openbmc/issues/1161

anoo1, 2017-05-05:
How do we know it's a dup if we have no data?
Please add the serial port output from a power cycle next time this is seen, before reflashing the system.
Reopening, you can close if this doesn't recreate.

anoo1, 2017-05-05:
Closing. Reopen if this is seen again and attached the data mentioned above.

closed 2017-05-05 by anoo1