openbmc/openbmc #1550: Implement Software Object.Delete

anoo1, 2017-05-03:


gtmills, 2017-05-08:
- [ ] Remove activation object
- [ ] Remove verision and file name objects
- [ ] Delete the image from /tmp

lgon, 2017-07-13:
Status Update
July 13, 2017
--------------------
Changed Version into a template class to avoid component dependency per Adriana's code review comment, and pushed change out to Gerrit.  Today, will test new openpower-update-manager binary on Witherspoon system and test Object.Delete.

rfrandse, 2017-07-19:
<https://gerrit.openbmc-project.xyz/5363> item_updater: Implement erase method
Resolves  #1550 Implement Software Object.Delete for pnor updater and image manager

lgon, 2017-07-20:
GitHub Comment
++++++++++++++++++++++++++++++++++++++
TASK LIST

- [ ]     Remove activation object
- [ ]     Remove verision and file name objects
- [ ]     Delete the image from /tmp

#**STATUS**

July 19, 2017
Made changes according to code review, and fixed issue with unit test that was causing Jenking build to fail. 
Waiting for reviewers to review latest changes. https://gerrit.openbmc-project.xyz/#/c/5242/
No Blockers

++++++++++++++++++++++++++++++++++++++

closed 2017-07-26 by williamspatrick