openbmc/openbmc #1554: Implement Software ActivationProgress.interface

anoo1, 2017-05-03:


mtritz, 2017-06-01:
- [ ] implement redundancy priority interface - appears during activation and disappears afterwards
- [ ] implement activation progress interface
- [ ] implement Set method - how should this work?
    - [ ] call Set in between steps of activation?
    - [ ] separate process in the background that fills in the values? like a timer?

mtritz, 2017-06-06:
This has prerequisite https://github.com/openbmc/openbmc/issues/1716

mtritz, 2017-06-26:
Reworking this - instead of a timer, will arbitrarily set and get an integer property so as to better reflect both the duration and state of the activation process.

geissonator, 2017-07-05:
https://gerrit.openbmc-project.xyz/#/c/5018/

closed 2017-07-07 by williamspatrick