openbmc/openbmc #1556: Python GC Hung Due to Python Bug

ongchris, 2017-05-03:
In Python 2.7, using subprocess.Popen with multiple threads could cause the garbage collector to be disabled. The workaround is to use the subprocess32 module in Python 2.7. The issue is discussed here: http://bugs.python.org/issue1336

Has this change been considered before for OpenBMC?

williamspatrick, 2017-05-03:
The only two cases of using subprocess.Popen that I can find are these:
    https://github.com/openbmc/skeleton/blob/520f8b05742916b920f20ac916a0e20fb981ce56/pyflashbmc/bmc_update.py#L261
https://github.com/openbmc/phosphor-networkd/blob/968d203ef934d68ded7e026d38dc77835116dedd/netman.py#L232

We are in the process of deprecating both of these programs and replacing them with something else.

Was there a specific case you were concerned about?

gtmills, 2018-05-31:
@ongchris We have moved to python3. Can this be closed?

mdmillerii, 2018-05-31:
@gtmills are the two programs identified by Patrick replaced?

gtmills, 2018-05-31:
@mdmillerii No but we moved to python3 which doesn't have the issue.

gtmills, 2018-05-31:
I am mistaken, we are still on python2. when we move to python I will close. 
```
root@Witherspoon:~# python -V
Python 2.7.13
```

closed 2019-04-27 by stale[bot]