openbmc/openbmc #1559: IPMI FRU commands : MRW parser

dkodihal, 2017-05-04:
Write new (or tweak existing IPMI inventory parser) MRW parser to help map IPMI FRU sensor record to a d-bus object path and d-bus properties. Output of the parser should be in YAML format.

devenrao, 2017-05-05:
- [ ]  Modify MRW parse script to generate YAML containing information about the
   IPMI Fru sensor record

- [ ] Map IPMI Fru sensor record to a d-bus object path and d-bus properties

devenrao, 2017-05-18:
- [ ] Consume the YAML file which has the IPMI mapping from FRU sensor record to dbus-object path and dbus-properties and generate C++ code.

- C++ code generated will aid in responding to IPMI FRU related commands.

geissonator, 2017-05-18:
Need task for recipe updates

rfrandse, 2017-06-01:
<https://gerrit.openbmc-project.xyz/4472> cater for generting fru inventory yaml file
Resolves  #1559 IPMI FRU commands : MRW parser

rfrandse, 2017-06-28:
<https://gerrit.openbmc-project.xyz/4657> Add native recipe to generate inventory map from MRW
Resolves  #1559 IPMI FRU commands : MRW parser

closed 2017-07-05 by williamspatrick