openbmc/openbmc #1560: IPMI FRU commands : implementation

dkodihal, 2017-05-04:
Consume YAML from #1559 via a python script, which in turn can C++ generate code that will aid responding to IPMI FRU related commands. This story if for the python script as well as the C++ code for implementing the commands.

devenrao, 2017-05-18:
- [ ] Based on FRU ID passed in locate the DBus object from Inventory

- [ ] Read the data from the DBus and fill in a local buffer as per the format defined

- [ ] Based on the offset and length specified read the data from the buffer package it and return to the caller.

geissonator, 2017-05-18:
If 2 points, need a more granular task break down.

devenrao, 2017-07-13:
Worked on netipmid not receiving PropertyChange events, tested the same and pushed the code for review.
Worked on task list for the new sprint
Reviewed code from Bard & Jayanth.

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5182> cater for property change signal to invalidate fru data
Resolves  #1560 IPMI FRU commands : implementation

closed 2017-07-27 by williamspatrick