openbmc/openbmc #1561: Request for Delete "all" Event Log command

swanman, 2017-05-04:
Adriana requested that I open an issue for this ... here is our chat:

swanman@us.ibm.com - Stephanie M Swanson/Rochester/IBM: 
10:10:54 AM: I hear there is a method to delete the event logs now on openbmc ... what is the command to do this? 

Adriana Kobylak: 
10:11:19 AM: let me get the rest format  

swanman@us.ibm.com - Stephanie M Swanson/Rochester/IBM: 
10:13:03 AM: thanks 

Adriana Kobylak: 
10:16:32 AM: curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST     -d '{"data": []}' https://<BMC>/xyz/openbmc_project/logging/entry/<ENTRYID>/action/Delethttps:///xyz/openbmc_project/logging/entry/2/action/Delete
where <BMC> is the bmc IP  and <ENTRYID> is the error log id like 1, 2, etc  

swanman@us.ibm.com - Stephanie M Swanson/Rochester/IBM: 
10:16:54 AM: Is there any way to delete all with a single command? 

Adriana Kobylak: 
10:18:14 AM: hmm there's not, that's probably something that would need to be added .. would you open an issue for it, should be straigh-fwd to loop through the events and delete each

stewart-ibm, 2017-08-07:
Having to cycle through each error log to delete it is rather painful, and added a non trivial amount of time to the execution of OPAL test suite.

whowutwut, 2017-08-11:
I second @stewart-ibm ....

Yes, it's a simple loop, but we have to first query, or list all the events, then put those into a list and for each item, run a delete command...  how many messages are allowed to be saved in the eventlog and what is the performance of running a loop through this? 

If there is a decision about providing an API to clear all can we document this so we can code against the missing endpoint and just fail until it's implemented.

ngorugan, 2017-08-24:
- [ ] Add DeleteAll method into xyz/openbmc_project/Logging/Internal/Manage.Interface.yaml

- [ ] Implement DeleteAll method

- [ ] Add test case(s) to validate.

xuweibj, 2017-08-24:
Hi @ngorugan , could you help to paste the `curl` command of the DeleleAll?

dkodihal, 2017-08-24:
@xuweibj we'll post that in day or so, as soon as the interface is finalized.

dkodihal, 2017-08-24:
@ngorugan your first task needs to change. This can't be in an internal interface. It should be in a new external interface.

ngorugan, 2017-08-24:
@dkodihal  I have updated, can you have look.

dkodihal, 2017-08-24:
Add a new interface Delete.interface.yaml at /xyz/openbmc_project/Logging/ ->  Add a new interface Delete.interface.yaml at /xyz/openbmc_project/Logging/ with a method DeleteAll, which is suppose to delete all committed errors. @ngorugan.

ngorugan, 2017-08-31:
This is how the yaml file looks like:

https://gerrit.openbmc-project.xyz/#/c/6313/

ngorugan, 2017-09-06:
yaml file got changed @xuweibj , Please have a look at below given link.
https://gerrit.openbmc-project.xyz/#/c/6397/

ngorugan, 2017-09-06:
# **STATUS**
09/06
Cod is under review

rfrandse, 2017-09-06:
<https://gerrit.openbmc-project.xyz/6314> Add implementation for delete all error log entries in one shot.
Resolves  #1561 Request for Delete "all" Event Log command

closed 2017-09-13 by williamspatrick

whowutwut, 2017-09-20:
Am I missing something? How come there is a disconnect when the issue is closed and when the dbus interfaces is updated? When does https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Logging show the new yaml file changes?

ngorugan, 2017-09-20:
yaml file is here : https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Collection/DeleteAll.interface.yaml

whowutwut, 2017-09-25:
@ngorugan  Thanks.   

Not directly at this, but this example falls into a struggle that we have been dealing with from the beginning.  When we look at the interfaces being defined for the API, there's a few questions that can't seem to be answered unless we physically run the code and see what is implemented. 

1. How do I call this? 
```
curl .... -X DELETE https://$IP/xyz/openbmc_project/Collection/DeleteAll? 
```
I don't think so.. it's a collection? What does the collection even look like?  

2. There's no README.md for the Logging endpoint: https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Logging

Looking at the latest FW, I see  `/xyz/openbmc_project/logging/internal/manager`?  There? 

How do we proactively write code against these APIs without waiting for the physical code to show up on the BMC?  Copying @bradbishop

bjwyman, 2017-11-29:
@whowutwut The phosphor-logging README.md has an example.
https://github.com/openbmc/phosphor-logging/blob/master/README.md
```
curl -c cjar -k -X POST -H "Content-Type: application/json" \
-d '{"data": [ "root", "<root password>" ] }' https://$BMC IP/login

curl -c cjar -b cjar -k -H "Content-Type: application/json" \
-X POST https://$BMC IP/xyz/openbmc_project/logging/action/deleteAll \
-d "{\"data\": [] }"
```