openbmc/openbmc #1563: Unable to set the fan speed

trajeswaran, 2017-05-04:
If I try to set the fan speed using the below curl command, I am getting an error.  Please find below the error message. 

curl command:
==========

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://w34.aus.stglabs.ibm.com/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT https://w34.aus.stglabs.ibm.com/xyz/openbmc_project/sensors/fan_tach/fan3/attr/Target -d "{\"data\" : 8300}"

Output:
=====
{
  "data": "User 'root' logged in", 
  "message": "200 OK", 
  "status": "ok"
}{
"data": {
    "description": "Internal Server Error", 
    "exception": "AttributeError(\"'NoneType' object has no attribute 'lower'\",)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
      "    return route.call(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 618, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 675, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 853, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 827, in wrap", 
      "    return self.to_jsonp(callback(*a, **kw))", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
      "    rv = callback(*a, **ka)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 792, in wrap", 
      "    data = callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 726, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 103, in __call__", 
      "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 309, in do_put", 
      "    path, prop, request.route_data['map'][path])", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 326, in get_host_interface", 
      "    path, prop, properties_iface, bus, interfaces)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 337, in get_host_interface_on_bus", 
      "    prop, properties.keys())", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/utils/misc.py\", line 41, in find_case_insensitive", 
      "    return next((x for x in lst if x.lower() == value.lower()), None)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/utils/misc.py\", line 41, in <genexpr>", 
      "    return next((x for x in lst if x.lower() == value.lower()), None)", 
      "AttributeError: 'NoneType' object has no attribute 'lower'"
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"

lgon, 2017-05-10:
This issue is related to openbmc/openbmc#1160, and is fixed by the code change for #1160.

lgon, 2017-05-12:
When testing my changes, I had to change the curl commands listed in the issue.  The double quotes around the {}'s had to be changed to single quotes.
Modified Curl Commands:
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://9.3.185.33/login -d '{"data": [ "root", "0penBmc" ] }'
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT https://9.3.185.33/xyz/openbmc_project/sensors/fan_tach/fan3/attr/Target -d '{"data" : 8300}'

rfrandse, 2017-05-17:
https://gerrit.openbmc-project.xyz/#/c/3743/
Fixes openbmc/openbmc#1563

closed 2017-05-26 by williamspatrick